Jakeler/ble-serial #2: Shutdown initiated right after start

augustozanellato, 2020-06-18:
When I try to connect to an HM-10 module using `sudo ble-serial -v -d 5c:f8:21:96:f0:c2` I ge tthe following output
```
23:15:30.70 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/1
23:15:31.187 | WARNING | __main__.py: Bluetooth connection lost
23:15:31.187 | WARNING | __main__.py: Shutdown initiated
23:15:31.188 | INFO | virtual_serial.py: Stopping UART thread
23:15:31.188 | INFO | __main__.py: Shutdown complete.
```

Jakeler, 2020-06-18:
That is obviously a bluetooth connection problem. 
Adapter not powered on? Maybe wrong address?  Already connected to something else?

btw: You don't need root for the serial, just for scanning.

augustozanellato, 2020-06-22:
> Adapter not powered on? 

The adapter is on, rfkill says it's unblocked, Gnome also says that Bluetooth is enabled
> Maybe wrong address?

It was copy-pasted from `ble-scan` output
> Already connected to something else?

Nope

Jakeler, 2020-06-22:
Hmm, in theory this should work. ble-serial just passes the address to the bluepy library and this log indicates that it throws a BTLEDisconnectError almost immediately, so possibly a bug in the library...

Does it work with `bluetoothctl` and `connect 5c:f8:21:96:f0:c2`? Just to rule out issue with the host setup.