Jakeler/ble-serial #4: Connection hangs if device uses address type other than public

AGlass0fMilk, 2020-06-19:
Hi @Jakeler! Thanks for making this project. It's been something I've wanted to do for a while and I'm surprised it hadn't been done until now! Great work.

Anyway, I was trying to use this with Nordic Semi's nRF52 UART service example and I ran into an issue. The current utility does not allow one to specify the BLE address _type_; it always defaults to `public`. In many cases, devices may advertise with a different address, e.g. `random static`, `random resolvable`.

The example I was using advertises with a `random static` address type and when I try to connect with this utility it hangs trying to connect.

I was able to step debug the code and found where it was hanging. It was somewhere inside the pyblue library where it sets up the BLE connection. It was here in the bluepy sources:

https://github.com/edwios/bluepy/blob/10f1cee90afb416f139949b86b491e4cfa98c886/bluepy/btle.py#L407-L416

I changed the address type to match what my device had and then the connection worked.

Perhaps we should add an additional flag that lets the user specify different address types? There only appear to be two available (random and public):
https://github.com/edwios/bluepy/blob/10f1cee90afb416f139949b86b491e4cfa98c886/bluepy/btle.py#L31-L32

I'm wondering if bluepy supports BTLE privacy or pairing/encryption :thinking:

Jakeler, 2020-06-19:
Hi @AGlass0fMilk, yes that is a good idea. 
I have added the option to change the address type now:
```
  -t {public,random}, --address-type {public,random}
                        BLE address type, either public or random (default: public)
```
Let me know if it works, because I don't have a "random" device to test.

It is interesting what features are actually in the bluepy source, but not mentioned in the docs. For example the `setSecurityLevel` function with low, medium and high. Have you tried these? But i don't see pairing options, so I guess it is "just works" only.
Also there is `setMTU`, I will see if this can increase the packet size and performance.

In addition I am thinking about switching to [pygattlib](https://bitbucket.org/OscarAcena/pygattlib/src/default/), because it is more actively maintained. Did not really look into it so far though. Do you think it would support better security?

AGlass0fMilk, 2020-06-23:
@Jakeler I have tested the `-t` flag and it fixes my problem! Thanks for making that change.

I have not used the bluepy library at all beyond debugging this problem.

I'll have to look at pygattlib. BLE on Linux has been woefully unsupported beyond very specific uses (mice, etc)... especially in the python realm.

AGlass0fMilk, 2020-06-23:
Resolved by e0f789d9364e8fa47c156ed99b89adb04ae5c31d

closed 2020-06-23 by AGlass0fMilk