Jakeler/ble-serial #7: ble-scan fails with "bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)"

FreddieChopin, 2020-07-16:
Installed the project on Arch Linux with Python 3.8.2, as instructed in the README.md. When trying to run `ble-scan` as root this is what I get:

```
$ ble-scan
Traceback (most recent call last):
  File "/usr/bin/ble-scan", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.8/site-packages/ble_serial/scan.py", line 52, in main
    scan(args.sec, args.deep)
  File "/usr/lib/python3.8/site-packages/ble_serial/scan.py", line 16, in scan
    devices = scanner.scan(time)
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 895, in scan
    self.start(passive=passive)
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 840, in start
    self._mgmtCmd(self._cmd()+"end")
  File "/usr/lib/python3.8/site-packages/bluepy/btle.py", line 324, in _mgmtCmd
    raise BTLEManagementError("Failed to execute management command '%s'" % (cmd), rsp)
bluepy.btle.BTLEManagementError: Failed to execute management command 'scanend' (code: 13, error: Invalid Parameters)
```

Any ideas what may be wrong here?

Jakeler, 2020-07-16:
Maybe related to https://github.com/IanHarvey/bluepy/issues/150? 
In short: Have you tried turning it off and on again? ;) 
Seriously on some bluetooth adapters a power cycle for example with `bluetoothctl` helped if a device didn't want to connect. Haven't seen this scan error so far though...
I am also running it on Arch and the current package versions, that should not cause issues.

FreddieChopin, 2020-07-16:
> In short: Have you tried turning it off and on again? ;)

I actually did (; I saw the issue you linked and tried this hint, but via `btmgmt power off` + `btmgmt power on` - I assume that's the same thing as mentioned there. This did not change anything.

My problem here is that `ble-scan` does not work at all, and `ble-serial` either disconnects instantly, or I need to give it an UUID value, in which case it does connect, but doesn't do anything else (the receiver and main UART loop are not started). The adapter I have is an Asus BT400, which has BCM20702A1 chip inside. I've manually added the required firmware, but the situation is generally the same with or without it. The adapter does work in the setup I'm trying to use here, because I managed to modify https://github.com/Gawhary/Qt-BLE-Tester project to connect to my device - I can send/receive data, so the hardware is most likely working fine. However I would really love to have a working tty port - that's why I'm here (;

Jakeler, 2020-07-17:
What device are you connecting to? Please also post the output of ble-serial started with UID and -v.

FreddieChopin, 2020-07-17:
> What device are you connecting to?

It's a Microchip RN4871 module. I think I know what may be the real issue here. This module does _NOT_ have any "read" characteristics, only 2 write characteristics...
[image]

> Please also post the output of ble-serial started with UID and -v.

Most of the time this:

```
$ ble-serial -d 80:1F:12:BB:DE:7E -w 49535343-fe7d-4ae5-8fa9-9fafd205e455 -v
17:05:56.801 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/4
17:05:58.395 | WARNING | __main__.py: Bluetooth connection failed
17:05:58.395 | WARNING | __main__.py: Shutdown initiated
17:05:58.395 | INFO | virtual_serial.py: Stopping UART thread
17:05:58.395 | INFO | __main__.py: Shutdown complete.
```

Sometimes this (I have to interrupt it, it just sits in the second line forever):

```
$ ble-serial -d 80:1F:12:BB:DE:7E -w 49535343-fe7d-4ae5-8fa9-9fafd205e455 -v
17:05:59.622 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/4
17:06:04.441 | INFO | interface.py: Connected device 80:1F:12:BB:DE:7E
^C17:07:00.533 | INFO | __main__.py: Keyboard interrupt received
17:07:00.533 | WARNING | __main__.py: Shutdown initiated
17:07:00.533 | INFO | virtual_serial.py: Stopping UART thread
17:07:00.533 | INFO | __main__.py: Shutdown complete.
```

Jakeler, 2020-07-17:
Okay, interesting. The characteristic setup is different, but it should be fine. ble-serial never explicitly reads, it just waits for notifications and handles them. UART TX has the notify property as expected. Write to UART RX.
Don't know why they also enable write to TX though.

The log indicates that it hangs while getting the characteristics from the device. If that would return anything you would either get an error (if the UUID is wrong) or it would proceed with the startup. So this is in both cases a connection problem, mostly likely caused by the host...

Unfortunately this all happens inside the bluepy library, so I can't do much. Just thinking of switching to some other lib (#5)
Can you try if scanning and listing works with them and your setup? Especially examples with [bleak](https://github.com/hbldh/bleak/tree/master/examples) or [bluezero](https://github.com/ukBaz/python-bluezero/tree/master/examples).

Jakeler, 2021-04-14:
I am closing this because I switched from bluepy to bleak now (#17). Feel free to open another issue if it still does not work.

closed 2021-04-14 by Jakeler