Jakeler/ble-serial #11: Additional configuration options and automated connections

AGlass0fMilk, 2020-11-03:
Hi!

I think in order for this tool to be more useful, there needs to be more configuration allowed. For example, connection parameters, such as connection interval and MTU, can severely affect latency. Configuring these in a generic manner will likely break support for many BLE modules. Therefore, these things should be able to be specified in a succinct way.

Further, in order to achieve more "USB to UART"/FTDI-like convenience, there should be a way to automate connections. This should be simple to achieve in Linux. For Windows, perhaps we could make a service that runs at startup and automatically connects to previously-connected BLE devices using [`pywin32`](http://thepythoncorner.com/dev/how-to-create-a-windows-service-in-python/)?

These automated connections and device-specific connection parameters could be configured using a `.ini` or `.json` file.

@Jakeler thoughts?

Jakeler, 2020-11-03:
Yes, that is a good idea, if we are adding more options config files would be useful, instead of too many parameters. Still ble-serial should contain sane defaults, so it is usable without configuration as well, ideally only requiring a device ID. The config file could then be specified with a optional `--config PATH` parameter.

For automated connections I think anyone can write a small script (at least on Linux) if required. This is more a topic for distro specific packaging, not necessarily for the tool itself. For example `pip` can't properly install system services etc. (outside site-packages). We could add a example in the readme or wiki though.
About Windows: AFAIK the `pty.openpty()`/`os.openpty()` functions used for the serial work only on Linux, so there is no Windows support in sight. I am not really into Windows anyway, that makes it low priority for me. But I would be happy to support it, if it does not require huge changes. Then I would take the same approach as with Linux, providing a example/service setup in the documentation.

Jakeler, 2021-05-29:
As far as I am aware ble-serial includes now all parameters that `bleak` exposes. Do you see anything else left? @AGlass0fMilk 
Connection interval seems to be not available though (https://github.com/hbldh/bleak/issues/149).

I have made script to automate connections: https://github.com/Jakeler/ble-serial/blob/master/helper/ble-autoconnect.py
It reads a INI config file, for example: https://github.com/Jakeler/ble-serial/blob/master/helper/autoconnect.ini
The config keys get directly transformed to long options, so `mtu = 20` -> `--mtu 20`.

commit 633ede7c (2021-06-02): Readme: mention ble-autoconnect
(and improve closing remarks)
closes #11
closed 2021-06-02 by Jakeler