Jakeler/ble-serial #12: No received data

coelner, 2021-01-07:
backlog: https://github.com/Jakeler/UT61E-Toolkit/issues/10
```
 [18:04|coelner@localhost ~]$ ble-scan -d
Discovered device: 6b:9e:58:2d:35:39 -> None
Discovered device: a4:c1:38:02:b9:dd -> UT61E
Discovered device: 0c:f3:ee:31:20:17 -> None
Found 3 devices!

Device 6b:9e:58:2d:35:39 (random), RSSI=-63 dB
    03: Complete 16b Services = 0000fd6f-0000-1000-8000-00805f9b34fb
    16: 16b Service Data = 6ffd94fd335da2744bb764f79d35a2fb85195e08c6a4
Could not read device: Failed to connect to peripheral 6b:9e:58:2d:35:39, addr type: random

Device a4:c1:38:02:b9:dd (public), RSSI=-79 dB
    01: Flags = 06
    02: Incomplete 16b Services = 0000ffe0-0000-1000-8000-00805f9b34fb,0000fee7-0000-1000-8000-00805f9b34fb
    ff: Manufacturer = cf9888a0a4c13802b9dd
    09: Complete Local Name = UT61E
  Service: 00001800-0000-1000-8000-00805f9b34fb
    Characteristic: 00002a00-0000-1000-8000-00805f9b34fb READ NOTIFY 
    Characteristic: 00002a01-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a04-0000-1000-8000-00805f9b34fb READ 
  Service: 0000ffe0-0000-1000-8000-00805f9b34fb
    Characteristic: 0000ffe1-0000-1000-8000-00805f9b34fb WRITE NO RESPONSE NOTIFY 
    Characteristic: 0000ffe2-0000-1000-8000-00805f9b34fb WRITE NO RESPONSE NOTIFY 

Device 0c:f3:ee:31:20:17 (public), RSSI=-66 dB
  Service: 00001800-0000-1000-8000-00805f9b34fb
    Characteristic: 00002a00-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a01-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002aa6-0000-1000-8000-00805f9b34fb READ 
  Service: 00001801-0000-1000-8000-00805f9b34fb
    Characteristic: 00002a05-0000-1000-8000-00805f9b34fb INDICATE 
  Service: 0000180d-0000-1000-8000-00805f9b34fb
    Characteristic: 00002a37-0000-1000-8000-00805f9b34fb NOTIFY 
    Characteristic: 00002a38-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a39-0000-1000-8000-00805f9b34fb WRITE 
  Service: 0000180a-0000-1000-8000-00805f9b34fb
    Characteristic: 00002a29-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a23-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a24-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a25-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a26-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a27-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a28-0000-1000-8000-00805f9b34fb READ 
    Characteristic: 00002a2a-0000-1000-8000-00805f9b34fb READ 
  Service: 00002760-08c2-11e1-9073-0e8ac72e1011
    Characteristic: 00002760-08c2-11e1-9073-0e8ac72e0011 WRITE NO RESPONSE 
    Characteristic: 00002760-08c2-11e1-9073-0e8ac72e0012 NOTIFY 
    Characteristic: 00002760-08c2-11e1-9073-0e8ac72e0013 WRITE NO RESPONSE NOTIFY 

✔ [18:05|coelner@localhost ~]$ ble-serial -v -d a4:c1:38:02:b9:dd
18:05:53.471 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/4
18:05:54.125 | INFO | interface.py: Connected device a4:c1:38:02:b9:dd
18:05:54.125 | DEBUG | interface.py: No write uuid specified, trying ['0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb']
18:05:54.484 | DEBUG | interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb
18:05:54.485 | INFO | interface.py: Receiver set up
18:05:54.485 | INFO | __main__.py: Running main loop!
18:06:59.553 | WARNING | __main__.py: Bluetooth connection failed
18:06:59.553 | WARNING | __main__.py: Shutdown initiated
18:06:59.553 | INFO | virtual_serial.py: Stopping UART thread
18:07:00.556 | INFO | interface.py: BT disconnected
18:07:00.556 | INFO | __main__.py: Shutdown complete.
✔ [18:07|coelner@localhost ~]$ 
```
After a minute I shut down the UT61E.

Jakeler, 2021-01-08:
Please try: `ble-serial -v -d a4:c1:38:02:b9:dd -r 0000ffe1-0000-1000-8000-00805f9b34fb` 
(or with `0000ffe2...`)

coelner, 2021-01-08:
```
✔ [07:16|coelner@localhost ~]$ ble-serial -v -d a4:c1:38:02:b9:dd -r 0000ffe1-0000-1000-8000-00805f9b34fb
07:17:39.223 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/4
07:17:39.618 | INFO | interface.py: Connected device a4:c1:38:02:b9:dd
07:17:39.618 | DEBUG | interface.py: No write uuid specified, trying ['0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb']
07:17:39.917 | DEBUG | interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb
07:17:40.217 | DEBUG | interface.py: Found read characteristic 0000ffe1-0000-1000-8000-00805f9b34fb
07:18:09.378 | INFO | interface.py: Receiver set up
07:18:09.379 | INFO | __main__.py: Running main loop!
07:18:12.338 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:12.338 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:12.361 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:12.361 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:12.361 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:12.361 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:13.827 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:13.827 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:15.298 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:15.298 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:16.786 | DEBUG | interface.py: Received notify: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
07:18:16.786 | DEBUG | virtual_serial.py: Write: b'\xb0\xb0\xb0\xb0\xb0\xb0;\xb0\xb0\xb0\xba\xb0\r\x8a'
[snip]
^C07:21:03.717 | INFO | __main__.py: Keyboard interrupt received
07:21:03.717 | WARNING | __main__.py: Shutdown initiated
07:21:03.717 | INFO | virtual_serial.py: Stopping UART thread
07:21:03.839 | INFO | interface.py: BT disconnected
07:21:03.839 | INFO | __main__.py: Shutdown complete.
```

improvement: fix the millisecond time print to 3 decimals, that makes the output readable

coelner, 2021-01-08:
```
07:36|coelner@localhost ~]$ ble-serial -v -d a4:c1:38:02:b9:dd -r 0000ffe2-0000-1000-8000-00805f9b34fb
07:36:46.21 | INFO | virtual_serial.py: Slave created on /tmp/ttyBLE -> /dev/pts/4
07:36:47.117 | INFO | interface.py: Connected device a4:c1:38:02:b9:dd
07:36:47.117 | DEBUG | interface.py: No write uuid specified, trying ['0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb']
07:36:47.616 | DEBUG | interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb
07:36:48.17 | DEBUG | interface.py: Found read characteristic 0000ffe2-0000-1000-8000-00805f9b34fb
07:37:12.600 | WARNING | __main__.py: Bluetooth connection failed
07:37:12.601 | WARNING | __main__.py: Shutdown initiated
07:37:12.601 | INFO | virtual_serial.py: Stopping UART thread
07:37:12.601 | INFO | __main__.py: Shutdown complete.
```

Jakeler, 2021-01-08:
Okay, this is the expected behavior. It does not subscribe per default to notifications, as only few modules require this. You just have to specify the read UUID in this case, I have clarified this in the readme now.

I have also fixed the timestamp to 3 decimals.

closed 2021-01-08 by Jakeler