Jakeler/ble-serial #15: Bluepy won't install, sudo: ble-scan: command not found

JBrakefield, 2021-03-23:
I've tried everything to get this running on my Ubuntu (20.04) WSL (I also tried getting it to run in Cygwin with no luck). I had a lot of trouble installing that bluepy library specifically, and when I get to the step "Find bluepy-helper" it is simply not there. My ~/.local/lib/python3.8/site-packages has no folder at all for bluepy, even after it finally installed with no errors.

I tried testing ble-scan anyway, but it says the command is not found when using sudo.

Jakeler, 2021-03-23:
Did you run `pip install` as normal user or root? It might be in the system site packages instead, so `/usr/lib/python3.8/site-packages`. You can check the paths with `python -m site`. Also I would recommend [python virtualenv](https://docs.python.org/3/library/venv.html) to get a isolated environment.

That being said, even after fixing that it will most likely not work, because WSL does not support [direct hardware access](https://www.reddit.com/r/bashonubuntuonwindows/comments/efzj4w/does_wsl_2_support_bluetooth/) or bluetooth in general: https://github.com/microsoft/WSL/issues/242
Please try it with a native Linux installation or live CD/USB stick. Windows is currently not supported, this might change in the future after the migration to another bluetooth backend is done (ble-serial version 2.0+).

JBrakefield, 2021-03-23:
Thanks for the response. You are probably correct that I can't use this on my Windows machine yet. I'm also trying to learn Bleak in hopes that library works for me.

Jakeler, 2021-04-13:
I am closing this as WSL is not supported, but I am migrating to Bleak in #17 and just merged Windows support #16 in. Checkout the `bleak-backend` branch and please let me know if this works for you.
It requires native python 3.8 and [com0com]( https://sourceforge.net/projects/signed-drivers/files/com0com/v3.0/). Data is read/written to `\\\\.\\BLE`. You can use the included `ble-setup` script to correctly create a `BLE`<>`COM9` port pair.

closed 2021-04-13 by Jakeler