Jakeler/ble-serial #23: cannot write to BLE modul via Bluetooth

arohrbacher, 2021-05-08:
Hi,
I am having a weired problem with your ble-serial script.
I am running it on a Raspberry Pi 3A with Raspbian Buster/Python 3.7.3; the BLE module I have is a AT-09/HM10-clone. 
After ble-scan I get the MAC address for this module.
starting ble-serial:
$ ble-serial -d xx:xx:xx:xx:xx:xx (MAC address obfuscated)
21:44:50.190 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/pts/1
21:44:50.190 | INFO | ble_interface.py: Receiver set up
21:44:50.207 | INFO | ble_interface.py: Trying to connect with xx:xx:xx:xx:xx:xx
21:44:52.008 | INFO | ble_interface.py: Device xx:xx:xx:xx:xx:xx connected
21:44:52.166 | INFO | main.py: Running main loop!
so the Bluetooth connection is working.
I am then starting a terminal program on my raspi and connect to /dev/pts/1 -> this works without error
My module is connected to the serial port on an other computer and this is running a terminal program (with correct baud rate etc). 
When I type on the remote terminal program the characters are received on the raspi and displayed in its terminal program: OK
However, when I type in the raspi's terminal program, the remote terminal doesn't receive anything and the Bluetooth connection shuts down:
z21:51:29.277 | ERROR | main.py: Bluetooth connection failed: org.bluez.Error.NotPermitted
21:51:29.277 | WARNING | main.py: Shutdown initiated
21:51:29.279 | INFO | linux_pty.py: Serial reader and symlink removed
21:51:31.907 | WARNING | ble_interface.py: Device xx:xx:xx:xx:xx:xx disconnected
21:51:31.908 | INFO | linux_pty.py: Stopping serial event loop
21:51:31.910 | INFO | ble_interface.py: Stopping Bluetooth event loop
21:51:31.912 | INFO | ble_interface.py: Bluetooth disconnected
21:51:31.913 | INFO | main.py: Shutdown complete.

The first character I entered locally was a "z", but this "z" didn't show up remotely.
This also happens when I write directly from the console to /dev/pts/1 : no data is received by the remote terminal program and the Bluetooth connection shuts down...

Any idea what I am doing wrong?

Thanks,

arohrbacher

Jakeler, 2021-05-08:
> 21:51:29.277 | ERROR | main.py: Bluetooth connection failed: org.bluez.Error.NotPermitted

Looks like this https://github.com/hbldh/bleak/issues/512 issue? Using BlueZ 5.51 seems to fix it for several people.

I guess running as root does not solve it? That is the first thing I would try with permission problems.

arohrbacher, 2021-05-09:
OK, thanks!

Indeed running as root didn't solve the issue. However, after looking at the [hbldh/bleak#512](https://github.com/hbldh/bleak/issues/512) issue I installed Bluez 5.51 and this fixed the issue for me: characters typed in the local terminal program showed up remotely and the Bluetooth connection stayed on. Just for grins I then installed Bluez 5.58 (the latest I could find): this also worked and my Bluetooth connection was stable.
Anyway, for me the issue is solved: upshot is that one has to use a Bluez version >=5.51

Thanks for the useful scripts and your help!

arohrbacher

Jakeler, 2021-05-09:
Awesome! Closing because it is solved.

closed 2021-05-09 by Jakeler