Jakeler/ble-serial #29: I try ble-serial on win10, but fail. Why?

cjgu00, 2021-05-26:
Hi!

I am really interested in this utility. It should be very useful. I try ble-serial on win10, but fail. Why?
[image]

Please help me... Thank you!

==============================================================
D:\WPy64-3880\scripts>ble-scan
Started BLE scan

C1:23:48:27:0A:32 (RSSI=-56): C_Watch
51:25:6E:D6:6F:62 (RSSI=-95):

Finished BLE scan

D:\WPy64-3880\scripts>ble-serial -d C1:23:48:27:0A:32
16:34:00.060 | INFO | ble_interface.py: Receiver set up
16:34:00.061 | ERROR | main.py: Unexpected Error: could not open port '\\\\.\\BLE': FileNotFoundError(2, '系統找不到指定的檔案。', None, 2)
16:34:00.061 | WARNING | main.py: Shutdown initiated
Traceback (most recent call last):
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\WPy64-3880\python-3.8.8.amd64\Scripts\ble-serial.exe\__main__.py", line 7, in <module>
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\site-packages\ble_serial\main.py", line 96, in launch
    m.start()
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\site-packages\ble_serial\main.py", line 13, in start
    asyncio.run(self._run())
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\site-packages\ble_serial\main.py", line 77, in _run
    self.uart.remove()
  File "d:\wpy64-3880\python-3.8.8.amd64\lib\site-packages\ble_serial\ports\windows_com0com.py", line 58, in remove
    self.serial.close()
AttributeError: 'COM' object has no attribute 'serial'
==============================================================

cjgu00, 2021-05-26:
> Hi @cjgu00 <https://github.com/cjgu00>,
> have you rebooted the system after the com0com setup?

Hi, Jake!

I DO reboot my notebook after installed and configured the com0com.
But it was the same anyway. Is there any others to setup or to config?

*C.J.*

cjgu00, 2021-05-27:
Hi, Jake!

I tried the same thing on another computer. It did work! So it should be the 
problem of my computer settings. Though I don't know what the problem is.
Sorry to bother you and thanks anyway!

C.J.

Jakeler, 2021-05-27:
Very nice! If you figure out sometime what setting it caused then please mention it here, to help others that might encounter the same issue. I mark this issue as closed now, but you can still answer.

closed 2021-05-27 by Jakeler

lautumn1990, 2021-07-06:
I have the same problem but I didn’t know how to solve it at the time. 
@cjgu00 @Jakeler Can you help me?

Jakeler, 2021-07-06:
@lautumn1990 
You could try a few things:
- run ble-serial with `-p COM9` (or any other port you have on the system)
- check if miniterm can connect and transfer data 
  - `python -m serial.tools.miniterm \\.\BLE`
  - `python -m serial.tools.miniterm COM9`
- check if the virtual ports work from other terminal programs, for example [ScriptCommunicator](https://sourceforge.net/projects/scriptcommunicator/)
- delete the port pair, reboot, set it up again, reboot
- change system settings, for example if you have set Windows to Chinese language try English (maybe it is somehow related to character encoding in paths)

Please report back what you find. If nothing works we could make a minimal example and open an issue upstream, though I probably can't help much because I am unable to reproduce this error...

cjgu00, 2021-07-07:
Hi!

Finally, I found why I failed on the 1st computer but succeeded on the 2nd one!
The 1st computer is a pretty new one of my friend. I missed the following note
because I had no idea what is 'secure boot'. Maybe some words about that could
be helpful for newbies like me.

[image]

After fixed, the new computer could run normally.

The 2nd computer is my old computer. It didn't have 'secure boot' option.
So it could run without changing any settings.

Hope this can help!

lautumn1990, 2021-07-07:
Great, it took effect after disable the `secure boot` option. I also missed this note. I thought it was not relevant. When I changed the settings from the BIOS, everything was normal.
@cjgu00 @Jakeler  Thanks a million!

bobmorane06, 2021-11-13:
If anyone has a similar issue, in my case it seems that it was the `ble-com-setup.exe` not working. After I manually set a pair in Com0Com and named a device BLE, it would work