Jakeler/ble-serial #31: Pythonnet dependency is not supported by Python 3.9.5

kylemccreery, 2021-05-28:
**Describe the bug**
Pip install fails from dependencies (pythonnet) on python 3.9.5. The purpose of this issue is to specify the versioning on https://pypi.org/project/ble-serial/

**Log messages**
```
> pip install ble-serial
Collecting ble-serial
  Downloading ble-serial-2.2.0.tar.gz (19 kB)
Collecting bleak>=0.11.0
  Downloading bleak-0.11.0-py2.py3-none-any.whl (108 kB)
     |████████████████████████████████| 108 kB 1.7 MB/s
Collecting coloredlogs>=15.0
  Downloading coloredlogs-15.0-py2.py3-none-any.whl (45 kB)
     |████████████████████████████████| 45 kB 1.0 MB/s
Requirement already satisfied: pyserial>=3.4.0 in c:\users\mccreerykr\appdata\local\programs\python\python39\lib\site-packages (from ble-serial) (3.5)
Collecting pythonnet
  Downloading pythonnet-2.5.2.tar.gz (1.9 MB)
     |████████████████████████████████| 1.9 MB 3.3 MB/s
Collecting humanfriendly>=9.1
  Downloading humanfriendly-9.1-py2.py3-none-any.whl (86 kB)
     |████████████████████████████████| 86 kB 1.9 MB/s
Collecting pyreadline
  Downloading pyreadline-2.1.zip (109 kB)
     |████████████████████████████████| 109 kB 3.3 MB/s
Requirement already satisfied: pycparser in c:\users\mccreerykr\appdata\local\programs\python\python39\lib\site-packages (from pythonnet->bleak>=0.11.0->ble-serial) (2.20)
Building wheels for collected packages: ble-serial, pyreadline, pythonnet
  Building wheel for ble-serial (setup.py) ... done
  Created wheel for ble-serial: filename=ble_serial-2.2.0-py3-none-any.whl size=18409 sha256=0b0533259e1407ed606118cb6df093b780bbbd4568ae3976c6c97e5d00ec592d
  Stored in directory: c:\users\mccreerykr\appdata\local\pip\cache\wheels\54\f6\56\9e2b107ad4b810197e98f0b8f7196beac4d7e1706fc3db892c
  Building wheel for pyreadline (setup.py) ... done
  Created wheel for pyreadline: filename=pyreadline-2.1-py3-none-any.whl size=93837 sha256=6ccd58dd095ae31fa4eff4e99e477992447fc023bce3d52a19a0568ee80bc66d
  Stored in directory: c:\users\mccreerykr\appdata\local\pip\cache\wheels\4d\c9\42\42ce63bbead3b4553087f8ad39f625586323bbac38f67f8d0d
  Building wheel for pythonnet (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\users\mccreerykr\appdata\local\programs\python\python39\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\mccreerykr\\AppData\\Local\\Temp\\pip-install-npvo4f75\\pythonnet_0a02a8f04d0b486386c9838e137527be\\setup.py'"'"'; __file__='"'"'C:\\Users\\mccreerykr\\AppData\\Local\\Temp\\pip-install-npvo4f75\\pythonnet_0a02a8f04d0b486386c9838e137527be\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\mccreerykr\AppData\Local\Temp\pip-wheel-dzs8j669'
       cwd: C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\
  Complete output (50 lines):
  running bdist_wheel
  running build
  running build_ext
  Checking for updates from https://www.nuget.org/api/v2/.
  Currently running NuGet.exe 4.1.0.
  Updating NuGet.exe to 5.9.1.
  Update successful.
  Cannot find the specified version of msbuild: '14'
  MSBuild auto-detection: using msbuild version '4.0' from 'C:\Windows\Microsoft.NET\Framework64\v4.0.30319'.
  C:\Users\mccreerykr\AppData\Local\Temp\NuGetScratch\2adsqviy.f22.nugetrestore.targets(460,20): error MSB4086: A numeric comparison was attempted on "$(MSBuildAssemblyVersion)" that evaluates to "" instead of a number, in condition "'$(MSBuildAssemblyVersion)' < '15.0'". [C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\src\runtime\Python.Runtime.csproj]
  C:\Users\mccreerykr\AppData\Local\Temp\NuGetScratch\2adsqviy.f22.nugetrestore.targets(460,20): error MSB4086: A numeric comparison was attempted on "$(MSBuildAssemblyVersion)" that evaluates to "" instead of a number, in condition "'$(MSBuildAssemblyVersion)' < '15.0'". [C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\src\testing\Python.Test.csproj]
  C:\Users\mccreerykr\AppData\Local\Temp\NuGetScratch\2adsqviy.f22.nugetrestore.targets(460,20): error MSB4086: A numeric comparison was attempted on "$(MSBuildAssemblyVersion)" that evaluates to "" instead of a number, in condition "'$(MSBuildAssemblyVersion)' < '15.0'". [C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\src\embed_tests\Python.EmbeddingTest.csproj]
  C:\Users\mccreerykr\AppData\Local\Temp\NuGetScratch\2adsqviy.f22.nugetrestore.targets(460,20): error MSB4086: A numeric comparison was attempted on "$(MSBuildAssemblyVersion)" that evaluates to "" instead of a number, in condition "'$(MSBuildAssemblyVersion)' < '15.0'". [C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\src\console\Console.csproj]
  C:\Users\mccreerykr\AppData\Local\Temp\NuGetScratch\2adsqviy.f22.nugetrestore.targets(460,20): error MSB4086: A numeric comparison was attempted on "$(MSBuildAssemblyVersion)" that evaluates to "" instead of a number, in condition "'$(MSBuildAssemblyVersion)' < '15.0'". [C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\src\clrmodule\clrmodule.csproj]

  WARNING: Error reading msbuild project information, ensure that your input solution or project file is valid. NETCore and UAP projects will be skipped, only packages.config files will be restored.
  Restoring NuGet package UnmanagedExports.1.2.7.
  Restoring NuGet package NUnit.ConsoleRunner.3.11.1.
  Restoring NuGet package System.ValueTuple.4.5.0.
  Restoring NuGet package NUnit.3.12.0.
    GET https://api.nuget.org/v3-flatcontainer/unmanagedexports/1.2.7/unmanagedexports.1.2.7.nupkg
    GET https://api.nuget.org/v3-flatcontainer/nunit/3.12.0/nunit.3.12.0.nupkg
    GET https://api.nuget.org/v3-flatcontainer/system.valuetuple/4.5.0/system.valuetuple.4.5.0.nupkg
    GET https://api.nuget.org/v3-flatcontainer/nunit.consolerunner/3.11.1/nunit.consolerunner.3.11.1.nupkg
    OK https://api.nuget.org/v3-flatcontainer/unmanagedexports/1.2.7/unmanagedexports.1.2.7.nupkg 52ms
    OK https://api.nuget.org/v3-flatcontainer/system.valuetuple/4.5.0/system.valuetuple.4.5.0.nupkg 220ms
    OK https://api.nuget.org/v3-flatcontainer/nunit/3.12.0/nunit.3.12.0.nupkg 221ms
    OK https://api.nuget.org/v3-flatcontainer/nunit.consolerunner/3.11.1/nunit.consolerunner.3.11.1.nupkg 225ms
  Installed UnmanagedExports 1.2.7 from https://api.nuget.org/v3/index.json with content hash bPIbzFaAj4AFxjHGFGPrYnUSCyIWJMkT/LFuiphWdpUbO2L9fP68W/1SP9cHCAxgiekV+kFNbybA2VvRo8PYIQ==.
  Installed NUnit.ConsoleRunner 3.11.1 from https://api.nuget.org/v3/index.json with content hash JONNeR6DlUMup+s9qTFxzdncsDlAtLbq45Cj6NiJJ8r3PzxOf93AviJ7jc0kFNQK9m7oSV5C/uR29KnePhm0BQ==.
  Adding package 'UnmanagedExports.1.2.7' to folder 'C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\packages'
  Adding package 'NUnit.ConsoleRunner.3.11.1' to folder 'C:\Users\mccreerykr\AppData\Local\Temp\pip-install-npvo4f75\pythonnet_0a02a8f04d0b486386c9838e137527be\packages'
  Installed System.ValueTuple 4.5.0 from https://api.nuget.org/v3/index.json with content hash okurQJO6NRE/apDIP23ajJ0hpiNmJ+f0BwOlB/cSqTLQlw5upkf+5+96+iG2Jw40G1fCVCyPz/FhIABUjMR+RQ==.
[... 62 lines cut ...]
plover                    4.0.0.dev8
plover-plugins-manager    0.5.16
plyer                     1.2.4
pycparser                 2.20
Pygments                  2.7.3
pyinstaller               4.1
pyinstaller-hooks-contrib 2020.11
pyparsing                 2.4.7
pyqt-distutils            0.7.3
PyQt5                     5.15.2
PyQt5-sip                 12.8.1
pyreadline                2.1
pyserial                  3.5
pywin32-ctypes            0.2.0
readme-renderer           28.0
requests                  2.25.1
requests-cache            0.5.2
requests-futures          1.0.0
setuptools                56.0.0
six                       1.15.0
urllib3                   1.26.2
wcwidth                   0.2.5
webencodings              0.5.1
wheel                     0.36.2
```

Jakeler, 2021-05-28:
`pythonnet` is not a direct dependency of `ble-serial`, but indirectly from `bleak` and only for Windows. 
So it works on Linux (I personally use always python 3.9). 
It should be possible to get it working on Windows as well like this: https://github.com/hbldh/bleak/discussions/509#discussioncomment-634621

That being said, as already mentioned in #32 I have specified the Python version to 3.7 and 3.8 now, this will be visible in the next release.

Jakeler, 2021-06-02:
Released in v2.2.1

Jakeler, 2021-07-06:
Bleak 0.12 got released, supporting Python 3.9 on Windows too, so the classifiers must be updated now.

commit 9ac62860 (2021-07-08): Bump bleak requierements version
supports py 3.9, closes #31
closed 2021-07-08 by Jakeler