Jakeler/ble-serial #36: Connecting to Arduino Nano 33 BLE

btoland, 2021-07-10:
Hi,

I might be completely on the wrong track here but I was trying to use your tool to connect to an Arduino Nano 33 BLE so that I could try read its accelerometer values via a serial port on my mac.

ble-scan can see it and reports back the read characteristics of the X and Y accelerations (what is published by the demo code on the Nano) but ble-serial will not connect as it reports back that there is no write characteristic to connect to.  See below.  I tried specifying the read characteristic UUID but it made no difference  - as expected from the -v output anyway.

Does this tool on work if the BLE hardware has a write characteristic?

Many thanks,
-Ben.

macos: Mojave 10.14.6 
Python version: 3.9.6

Started BLE scan
$ble-scan
D059B613-43A6-4285-A492-7F175C9C96EF (RSSI=-62): Arduino Accelerometer
20B0AD2A-34E9-40DF-B0F0-D7CA66315236 (RSSI=-64): Unknown
E6FA446F-A52A-4C7A-B142-47C292B60C01 (RSSI=-78): Unknown

Finished BLE scan

$ble-scan -d D059B613-43A6-4285-A492-7F175C9C96EF
Started deep scan of D059B613-43A6-4285-A492-7F175C9C96EF

SERVICE 00001101-0000-1000-8000-00805f9b34fb (Handle: 10): Serial Port
     CHARACTERISTIC 00002101-0000-1000-8000-00805f9b34fb (Handle: 11): Vendor specific ['read', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration
     CHARACTERISTIC 00002102-0000-1000-8000-00805f9b34fb (Handle: 14): Vendor specific ['read', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 16): Client Characteristic Configuration

Completed deep scan of D059B613-43A6-4285-A492-7F175C9C96EF

ble-serial -d D059B613-43A6-4285-A492-7F175C9C96EF
23:59:53.259 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/ttys002
23:59:53.259 | INFO | ble_interface.py: Receiver set up
23:59:53.259 | INFO | ble_interface.py: Trying to connect with D059B613-43A6-4285-A492-7F175C9C96EF
23:59:56.396 | INFO | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF connected
23:59:56.397 | ERROR | main.py: Unexpected Error: No characteristic with specified UUID ['0000ff01-0000-1000-8000-00805f9b34fb', '0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb'] found!
23:59:56.397 | WARNING | main.py: Shutdown initiated
23:59:56.397 | INFO | linux_pty.py: Serial reader and symlink removed
23:59:56.433 | WARNING | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF disconnected
23:59:56.434 | INFO | linux_pty.py: Stopping serial event loop
23:59:56.434 | INFO | ble_interface.py: Stopping Bluetooth event loop
23:59:56.434 | INFO | ble_interface.py: Bluetooth disconnected
23:59:56.434 | INFO | main.py: Shutdown complete.

$ ble-serial -d D059B613-43A6-4285-A492-7F175C9C96EF -v
00:02:23.400 | DEBUG | main.py: Running: Namespace(verbose=True, port='/tmp/ttyBLE', device='D059B613-43A6-4285-A492-7F175C9C96EF', timeout=5.0, addr_type='public', adapter='hci0', mtu=20, write_uuid=None, read_uuid=None, filename=None, binlog=False)
00:02:23.401 | DEBUG | selector_events.py: Using selector: KqueueSelector
00:02:23.402 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/ttys002
00:02:23.402 | INFO | ble_interface.py: Receiver set up
00:02:23.403 | INFO | ble_interface.py: Trying to connect with D059B613-43A6-4285-A492-7F175C9C96EF
00:02:25.896 | INFO | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF connected
00:02:25.896 | DEBUG | ble_interface.py: No write uuid specified, trying builtin list
00:02:25.896 | ERROR | main.py: Unexpected Error: No characteristic with specified UUID ['0000ff01-0000-1000-8000-00805f9b34fb', '0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb'] found!
00:02:25.896 | WARNING | main.py: Shutdown initiated
00:02:25.896 | INFO | linux_pty.py: Serial reader and symlink removed
00:02:25.933 | WARNING | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF disconnected
00:02:25.933 | DEBUG | main.py: Asyncio execption handler called D059B613-43A6-4285-A492-7F175C9C96EF disconnected!
00:02:25.934 | INFO | linux_pty.py: Stopping serial event loop
00:02:25.934 | INFO | ble_interface.py: Stopping Bluetooth event loop
00:02:25.934 | INFO | ble_interface.py: Bluetooth disconnected
00:02:25.934 | INFO | main.py: Shutdown complete.

ble-serial -d D059B613-43A6-4285-A492-7F175C9C96EF -r 00002101-0000-1000-8000-00805f9b34fb -v
00:28:59.162 | DEBUG | main.py: Running: Namespace(verbose=True, port='/tmp/ttyBLE', device='D059B613-43A6-4285-A492-7F175C9C96EF', timeout=5.0, addr_type='public', adapter='hci0', mtu=20, write_uuid=None, read_uuid='00002101-0000-1000-8000-00805f9b34fb', filename=None, binlog=False)
00:28:59.163 | DEBUG | selector_events.py: Using selector: KqueueSelector
00:28:59.165 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/ttys002
00:28:59.165 | INFO | ble_interface.py: Receiver set up
00:28:59.165 | INFO | ble_interface.py: Trying to connect with D059B613-43A6-4285-A492-7F175C9C96EF
00:29:03.538 | INFO | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF connected
00:29:03.538 | DEBUG | ble_interface.py: No write uuid specified, trying builtin list
00:29:03.538 | ERROR | main.py: Unexpected Error: No characteristic with specified UUID ['0000ff01-0000-1000-8000-00805f9b34fb', '0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb'] found!
00:29:03.538 | WARNING | main.py: Shutdown initiated
00:29:03.539 | INFO | linux_pty.py: Serial reader and symlink removed
00:29:03.574 | WARNING | ble_interface.py: Device D059B613-43A6-4285-A492-7F175C9C96EF disconnected
00:29:03.575 | DEBUG | main.py: Asyncio execption handler called D059B613-43A6-4285-A492-7F175C9C96EF disconnected!
00:29:03.575 | INFO | linux_pty.py: Stopping serial event loop
00:29:03.575 | INFO | ble_interface.py: Stopping Bluetooth event loop
00:29:03.576 | INFO | ble_interface.py: Bluetooth disconnected
00:29:03.576 | INFO | main.py: Shutdown complete.

Jakeler, 2021-07-11:
Hi Ben,
the tool always searches for a write characteristic, even if you don't specify one (the builtin list is used for that). It is currently not possible to disable this through parameters. So you have to either add a write characteristic to the Arduino or modify ble-serial. A makeshift solution would be to just comment out or remove [this line](https://github.com/Jakeler/ble-serial/blob/v2.3.0/ble_serial/bluetooth/ble_interface.py#L20). Then it should connect and work for reading, but crash if you write something to the UART. 
I will think about a proper fix, a 'read only' (or 'write only') option could be potentially useful, instead of always requiring both directions.

Though with reading you encounter another problem, because you want to read from 2 characteristics simultaneously?

btoland, 2021-07-11:
Hi Jake,

Cheers for the quick response.

I can easily add a dummy write characteristic to my code so that will be the solution I'll go for now.

I was only playing with demo code on the Nano last night and it split the accelerometer axises into separate read characteristics. It will be easy to put the X, y, and z readings into one characteristic so I'll run with that. 

If you get a chance to look into a read only solution it might be useful for people starting to play with the Nano as they could use your tool with the demo code out of the box. 

Again, many thanks for the reply and information.

Much appreciated,
-Ben.

Jakeler, 2021-07-13:
Done in [v2.4.0](https://github.com/Jakeler/ble-serial/releases/tag/v2.4.0)

closed 2021-07-13 by Jakeler