Jakeler/ble-serial #39: Unexpected Error: No characteristic with ['notify', 'indicate'] property found!

kalstate, 2021-08-22:
**Describe the bug**
Hello Jakeler,

Thanks for writing this code!

Issue: When I try connecting:
ble-serial -v -d 0F:82:5F:BF:B9:DE -w 13012f04-f8c3-4f4a-a8f4-15cd926da146 -r 13012f04-f8c3-4f4a-a8f4-15cd926da146

I get the following error:
Unexpected Error: No characteristic with ['notify', 'indicate'] property found!

In my arduino ino file, I've got: 
#include <ArduinoBLE.h>
BLEService nanoService("13012f00-f8c3-4f4a-a8f4-15cd926da146");
BLEIntCharacteristic servoCharacteristic("13012f04-f8c3-4f4a-a8f4-15cd926da146");

I've tried a few different approaches, but this one seems closest:

Thanks,
Chris

**Log messages**
pi@raspberrypi:~ $ ble-serial -v -d 0F:82:5F:BF:B9:DE -w 13012f04-f8c3-4f4a-a8f4-15cd926da146 -r 13012f04-f8c3-4f4a-a8f4-15cd926da146
08:59:27.251 | DEBUG | main.py: Running: Namespace(adapter='hci0', addr_type='public', binlog=False, device='0F:82:5F:BF:B9:DE', filename=None, mode='rw', mtu=20, port='/tmp/ttyBLE', read_uuid='13012f04-f8c3-4f4a-a8f4-15cd926da146', timeout=5.0, verbose=True, write_uuid='13012f04-f8c3-4f4a-a8f4-15cd926da146')
08:59:27.252 | DEBUG | selector_events.py: Using selector: EpollSelector
08:59:27.254 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/pts/1
08:59:27.254 | INFO | ble_interface.py: Receiver set up
08:59:27.270 | INFO | ble_interface.py: Trying to connect with 0F:82:5F:BF:B9:DE
08:59:28.323 | INFO | ble_interface.py: Device 0F:82:5F:BF:B9:DE connected
08:59:28.324 | DEBUG | ble_interface.py: Characteristic candidates for write: 
	13012f04-f8c3-4f4a-a8f4-15cd926da146 (Handle: 11): Unknown ['read', 'write']
08:59:28.325 | INFO | ble_interface.py: Found write characteristic 13012f04-f8c3-4f4a-a8f4-15cd926da146 (H. 11)
08:59:28.325 | DEBUG | ble_interface.py: Characteristic candidates for notify: 
	13012f04-f8c3-4f4a-a8f4-15cd926da146 (Handle: 11): Unknown ['read', 'write']
08:59:28.325 | ERROR | main.py: Unexpected Error: No characteristic with ['notify', 'indicate'] property found!
08:59:28.326 | WARNING | main.py: Shutdown initiated
08:59:28.327 | INFO | linux_pty.py: Serial reader and symlink removed
08:59:30.266 | WARNING | ble_interface.py: Device 0F:82:5F:BF:B9:DE disconnected
08:59:30.267 | DEBUG | main.py: Asyncio execption handler called 0F:82:5F:BF:B9:DE disconnected!
08:59:30.267 | INFO | linux_pty.py: Stopping serial event loop
08:59:30.267 | INFO | ble_interface.py: Stopping Bluetooth event loop
08:59:30.268 | INFO | ble_interface.py: Bluetooth disconnected
08:59:30.268 | INFO | main.py: Shutdown complete.

**Setup (please complete the following information):**
- OS: raspberry pi
- Bluetooth Hardware: arduino nano ble sense
- BlueZ Version: bluetoothctl: 5.50
- Python Version: Python 2.7.16
- ble-serial and dependency versions: 
- pi@raspberrypi:~ $ pip list
Package           Version    
----------------- -----------
arandr            0.1.9      
asn1crypto        0.24.0     
automationhat     0.2.0      
blinker           1.4        
blinkt            0.1.2      
buttonshim        0.0.2      
Cap1xxx           0.1.3      
certifi           2018.8.24  
chardet           3.0.4      
Click             7.0        
colorama          0.3.7      
colorzero         1.1        
configparser      3.5.0b2    
cookies           2.2.1      
cryptography      2.6.1      
drumhat           0.1.0      
entrypoints       0.3        
enum34            1.1.6      
envirophat        1.0.0      
ExplorerHAT       0.4.2      
Flask             1.0.2      
fourletterphat    0.1.0      
funcsigs          1.0.2      
gpiozero          1.6.2      
gyp               0.1        
idna              2.6        
ipaddress         1.0.17     
itsdangerous      0.24       
Jinja2            2.10       
keyring           17.1.1     
keyrings.alt      3.1.1      
MarkupSafe        1.1.0      
mcpi              0.1.1      
microdotphat      0.2.1      
mock              2.0.0      
mote              0.0.4      
motephat          0.0.3      
numpy             1.16.2     
oauthlib          2.1.0      
olefile           0.46       
pantilthat        0.0.7      
pbr               4.2.0      
phatbeat          0.1.1      
pianohat          0.1.0      
picamera          1.13       
picraft           1.0        
piglow            1.2.5      
pigpio            1.78       
Pillow            5.4.1      
pip               18.1       
pycairo           1.16.2     
pycrypto          2.6.1      
pyflakes          2.0.0      
pygame            1.9.4.post1
PyGObject         3.30.4     
pyinotify         0.9.6      
PyJWT             1.7.0      
pyOpenSSL         19.0.0     
pyserial          3.4        
pyxdg             0.25       
rainbowhat        0.1.0      
requests          2.21.0     
requests-oauthlib 1.0.0      
responses         0.9.0      
RPi.GPIO          0.7.0      
RTIMULib          7.2.1      
scrollphat        0.0.7      
scrollphathd      1.2.1      
SecretStorage     2.3.1      
sense-emu         1.1        
sense-hat         2.2.0      
setuptools        40.8.0     
simplejson        3.16.0     
six               1.12.0     
skywriter         0.0.7      
sn3218            1.2.7      
spidev            3.5        
touchphat         0.0.1      
twython           3.7.0      
unicornhathd      0.0.4      
urllib3           1.24.1     
Werkzeug          0.14.1     
wheel             0.32.3

Jakeler, 2021-08-22:
Hi @kalstate,
the log suggests that your `servoCharacteristic` has only `['read', 'write']` properties. It correctly starts the write module, but fails to setup the reader. ble-serial requieres the notify/indicate feature, to efficiently get informed when new data is available.
You just have to enable it in you Arduino code,  like this: 
```c
BLEIntCharacteristic servoCharacteristic("13012f04-f8c3-4f4a-a8f4-15cd926da146", BLENotify | BLEWrite | BLERead));
```
See the [Arduino reference doc](https://www.arduino.cc/en/Reference/ArduinoBLEBLECharacteristicBLECharacteristic). You can also check if it worked with `ble-scan`.

As a side note: Python Version 2.7 and your pip list can't be correct, you probably have to use `python3`/`pip3` to get the relevant info. 
But you obviously have a supported Python 3 interpreter installed, otherwise it would not work so far. So your Raspi installation should work fine if you adjust the Arduino code as described above.

kalstate, 2021-08-22:
> Hi @kalstate,
> the log suggests that your `servoCharacteristic` has only `['read', 'write']` properties. It correctly starts the write module, but fails to setup the reader. ble-serial requieres the notify/indicate feature, to efficiently get informed when new data is available.
> You just have to enable it in you Arduino code, like this:
> 
> ```c
> BLEIntCharacteristic servoCharacteristic("13012f04-f8c3-4f4a-a8f4-15cd926da146", BLENotify | BLEWrite | BLERead);
> ```
> 
> See the [Arduino reference doc](https://www.arduino.cc/en/Reference/ArduinoBLEBLECharacteristicBLECharacteristic). You can also check if it worked with `ble-scan`.
> 
> As a side note: Python Version 2.7 and your pip list can't be correct, you probably have to use `python3`/`pip3` to get the relevant info.
> But you obviously have a supported Python 3 interpreter installed, otherwise it would not work so far. So your Raspi installation should be fine if you adjust the Arduino code as described above.

Connected! Thanks!

closed 2021-08-22 by kalstate