Jakeler/ble-serial #44: ERROR | main.py: Unexpected Error:

XxGMTxX, 2021-09-23:
**Describe the bug**
When i try to connect i get the blank " Unexpected Error:" message.

**Log messages**
```
02:25:31.947 | DEBUG | main.py: Running: Namespace(verbose=True, port='BLE', device='30:45:11:F6:9B:43', timeout=30.0, addr_type='public', adapter='hci0', mtu=20, write_uuid=None, read_uuid=None, mode='rw', filename=None, binlog=False)
02:25:31.947 | DEBUG | proactor_events.py: Using proactor: IocpProactor
02:25:31.950 | INFO | ble_interface.py: Receiver set up
02:25:31.952 | INFO | ble_interface.py: Trying to connect with 30:45:11:F6:9B:43
02:25:32.762 | INFO | client.py: Services resolved for BleakClientWinRT (30:45:11:F6:9B:43)
02:25:32.762 | INFO | ble_interface.py: Device 30:45:11:F6:9B:43 connected
02:25:32.764 | DEBUG | ble_interface.py: No write uuid specified, trying builtin list
02:25:32.767 | DEBUG | ble_interface.py: Characteristic candidates for write:
        0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 17): www.jnhuamao.cn ['read', 'write-without-response', 'write', 'notify', 'indicate']
02:25:32.767 | INFO | ble_interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
02:25:32.768 | DEBUG | ble_interface.py: No notify uuid specified, trying builtin list
02:25:32.768 | DEBUG | ble_interface.py: Characteristic candidates for notify:
        0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 17): www.jnhuamao.cn ['read', 'write-without-response', 'write', 'notify', 'indicate']
02:25:32.769 | INFO | ble_interface.py: Found notify characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
02:25:33.041 | ERROR | main.py: Unexpected Error:
02:25:33.041 | WARNING | main.py: Shutdown initiated
02:25:36.240 | WARNING | ble_interface.py: Device 30:45:11:F6:9B:43 disconnected
02:25:36.242 | DEBUG | main.py: Asyncio execption handler called 30:45:11:F6:9B:43 disconnected!
02:25:36.244 | INFO | windows_com0com.py: Stopping RX+TX loop
02:25:36.247 | INFO | ble_interface.py: Stopping Bluetooth event loop
Traceback (most recent call last):
  File "C:\Python39\lib\asyncio\locks.py", line 226, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Python39\lib\asyncio\tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python39\Scripts\ble-serial-script.py", line 33, in <module>
    sys.exit(load_entry_point('ble-serial==2.4.1', 'console_scripts', 'ble-serial')())
  File "C:\Python39\lib\site-packages\ble_serial\main.py", line 103, in launch
    m.start()
  File "C:\Python39\lib\site-packages\ble_serial\main.py", line 13, in start
    asyncio.run(self._run())
  File "C:\Python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "C:\Python39\lib\site-packages\ble_serial\main.py", line 86, in _run
    await self.bt.disconnect()
  File "C:\Python39\lib\site-packages\ble_serial\bluetooth\ble_interface.py", line 96, in disconnect
    await self.dev.disconnect()
  File "C:\Python39\lib\site-packages\bleak\backends\winrt\client.py", line 262, in disconnect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Python39\lib\asyncio\tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

**Setup (please complete the following information):**
- OS: [Windows 10 pro build 19043.1237]
- Bluetooth Hardware: [Original HM-10 version: v710]
- Python Version: [Python 3.9.7]
- ble-serial and dependency versions:

```
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
Package       Version
------------- -----------
ble-serial    2.4.1
bleak         0.12.1
coloredlogs   15.0.1
humanfriendly 10.0
pip           21.2.3
pyreadline3   3.3
pyserial      3.5
setuptools    57.4.0
winrt         1.0.21033.1
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
WARNING: Ignoring invalid distribution -ip (c:\python39\lib\site-packages)
```

Jakeler, 2021-09-30:
First of all, sorry for the late response! 
An empty error is really hard to debug... I can't reproduce it so far with exactly the same Windows build and other software versions, so I need your assistance. If you can try different hardware combinations, it could be helpful to figure out what causes it. 
Could be the one characteristic with all properties on that HM-10 version, but this is just a guess. Maybe it also picks the wrong ID, can you post a deep scan?

Jakeler, 2021-09-30:
I have just tried an all in one characteristic on an ESP32:
```
SERVICE 0000ffe0-0000-1000-8000-00805f9b34fb (Handle: 40): Vendor specific
     CHARACTERISTIC 0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 41):  ['read', 'write-without-response', 'write', 'notify', 'indicate']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 43): Client Characteristic Configuration
```
which works fine, so at least that is not the issue:
```
[...]
21:00:55.960 | INFO | ble_interface.py: Trying to connect with ...
21:00:58.774 | INFO | client.py: Services resolved for BleakClientWinRT (...)
21:00:58.774 | INFO | ble_interface.py: Device ... connected
21:00:58.774 | DEBUG | ble_interface.py: No write uuid specified, trying builtin list
21:00:58.774 | DEBUG | ble_interface.py: Characteristic candidates for write:
        0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 41):  ['read', 'write-without-response', 'write', 'notify', 'indicate']
21:00:58.774 | INFO | ble_interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 41)
21:00:58.774 | DEBUG | ble_interface.py: No notify uuid specified, trying builtin list
21:00:58.774 | DEBUG | ble_interface.py: Characteristic candidates for notify:
        0000ffe1-0000-1000-8000-00805f9b34fb (Handle: 41):  ['read', 'write-without-response', 'write', 'notify', 'indicate']
21:00:58.774 | INFO | ble_interface.py: Found notify characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 41)
21:00:59.007 | INFO | main.py: Running main loop!
```