Jakeler/ble-serial #48: Unable to connect to nRF5240 using Adafruit Bluefruit BLEuart example

viggi1000, 2021-12-02:
**Describe the bug**
A clear and concise description of what the bug is.
Hi,
I am trying to use this tool with an nRF5240 running the nRF5-Arduino core using the Bluefruit library with the default BLEuart peripheral example. However, despite it using the same TX characteristic as the Nordic Uart Services (6E400003-B5A3-F393-E0A9-E50E24DCCA9E) it was unable to connect and suggested pairing the device. Even after pairing, I have been facing the same issue. 
I have tried a similar example with an ESP32 to great success with the same TX UART characteristic. I would like some assistance in identifying where the incompatibility is arising from and how I can resolve the same. I have verified the same example to be working as intended with the nrf5240 through both the Adafruit Bluefruit Connect app and nRF connect android app.

Thank you in advance!

**Log messages**

BLE Scan:

```
ble-scan -d C2:D0:E6:8D:EE:A2
Started deep scan of C2:D0:E6:8D:EE:A2

SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2):  ['read', 'write']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4):  ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6):  ['read']
     CHARACTERISTIC 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 8):  ['read']
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 10): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 11):  ['indicate']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 13): Client Characteristic Configuration
SERVICE 00001530-1212-efde-1523-785feabcd123 (Handle: 14): Device Firmware Update Service
     CHARACTERISTIC 00001532-1212-efde-1523-785feabcd123 (Handle: 15):  ['write-without-response']
     CHARACTERISTIC 00001531-1212-efde-1523-785feabcd123 (Handle: 17):  ['write', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 19): Client Characteristic Configuration
     CHARACTERISTIC 00001534-1212-efde-1523-785feabcd123 (Handle: 20):  ['read']
SERVICE 0000180a-0000-1000-8000-00805f9b34fb (Handle: 22): Device Information
     CHARACTERISTIC 00002a24-0000-1000-8000-00805f9b34fb (Handle: 23):  ['read']
     CHARACTERISTIC 00002a25-0000-1000-8000-00805f9b34fb (Handle: 25):  ['read']
     CHARACTERISTIC 00002a26-0000-1000-8000-00805f9b34fb (Handle: 27):  ['read']
     CHARACTERISTIC 00002a28-0000-1000-8000-00805f9b34fb (Handle: 29):  ['read']
     CHARACTERISTIC 00002a29-0000-1000-8000-00805f9b34fb (Handle: 31):  ['read']
SERVICE 6e400001-b5a3-f393-e0a9-e50e24dcca9e (Handle: 33): Nordic UART Service
     CHARACTERISTIC 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 34): TXD ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 36): Client Characteristic Configuration
         DESCRIPTOR 00002901-0000-1000-8000-00805f9b34fb (Handle: 37): Characteristic User Description
     CHARACTERISTIC 6e400002-b5a3-f393-e0a9-e50e24dcca9e (Handle: 38):  ['write-without-response', 'write']
         DESCRIPTOR 00002901-0000-1000-8000-00805f9b34fb (Handle: 40): Characteristic User Description
SERVICE 0000180f-0000-1000-8000-00805f9b34fb (Handle: 41): Battery Service
     CHARACTERISTIC 00002a19-0000-1000-8000-00805f9b34fb (Handle: 42):  ['read', 'notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 44): Client Characteristic Configuration

Completed deep scan of C2:D0:E6:8D:EE:A2

BLE Serial:
ble-serial -d C2:D0:E6:8D:EE:A2 -r 6e400003-b5a3-f393-e0a9-e50e24dcca9e
18:57:25.206 | INFO | ble_interface.py: Receiver set up
18:57:25.206 | INFO | ble_interface.py: Trying to connect with C2:D0:E6:8D:EE:A2
18:57:25.329 | ERROR | main.py: Bluetooth connection failed: Failed to connect to 214202477047458. If the device requires pairing, then pair first. If the device uses a random address, it may have changed.
18:57:25.329 | WARNING | main.py: Shutdown initiated
18:57:25.330 | INFO | main.py: Shutdown complete.
```

**Setup (please complete the following information):**
-OS: [Windows 10 21H1, com2com driver installed and configured]
- Bluetooth Hardware: [Realtek rtl8852ae]
- Python Version: [Python 3.8.8`]
- ble-serial and dependency versions: [ble-serial 2.4.1]

Jakeler, 2021-12-04:
Maybe it has a different address type? You can specify this with:
```
  -a {public,random}, --address-type {public,random}
                        BLE address type, either public or random (default: public)
```
I think it could be related to #4, because this was with the nRF52 UART service example as well.

Also please run ble-serial with `-v` to get a more verbose output, otherwise you can get even more (low level) logging from bleak like described here: https://github.com/hbldh/bleak/issues/647#issuecomment-934609138

whipple63, 2021-12-22:
I hope this is helpful.  I have been making successful connections with the Adafruit Bluefruit bleuart board as per the following:

[INFO] ['ble-serial', '--dev', 'E3:A0:40:C6:99:13', '--read-uuid', '6e400003-b5a3-f393-e0a9-e50e24dcca9e', '--write-uuid', '6e400002-b5a3-f393-e0a9-e50e24dcca9e', '--timeout', '10', '--mtu', '20', '--port', '/tmp/ttyBLE']
13:49:10.136 | INFO | linux_pty.py: Slave created on /tmp/ttyBLE -> /dev/pts/0
13:49:10.137 | INFO | ble_interface.py: Receiver set up
13:49:10.150 | INFO | ble_interface.py: Trying to connect with E3:A0:40:C6:99:13
13:49:11.588 | INFO | ble_interface.py: Device E3:A0:40:C6:99:13 connected
13:49:11.589 | INFO | ble_interface.py: Found write characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e (H. 36)
13:49:11.589 | INFO | ble_interface.py: Found notify characteristic 6e400003-b5a3-f393-e0a9-e50e24dcca9e (H. 32)
13:49:11.714 | INFO | main.py: Running main loop!

viggi1000, 2022-02-11:
Hi sorry for the late response, I just got back to bench and was waiting for my new Xiao NRF52 BLE boards to arrive before I tried this again. I will keep you posted once your suggestion! 

Thank you!

viggi1000, 2022-02-11:
I tried the command @whipple63 suggested, works perfectly with both the sparkfun NRF52 board and the Seeed Xiao NRF52!

Thank you for your assistance wondeful people! Just one quick question, would it be possible to create multiple COM port objects to subscribe to different GATT services or different BLE devices? I was unsure if I needed to run ble-com-setup.exe multiple times and how I could assign different COM ports with it.

Jakeler, 2022-02-11:
Nice, perfect!

> Just one quick question, would it be possible to create multiple COM port objects to subscribe to different GATT services or different BLE devices? I was unsure if I needed to run ble-com-setup.exe multiple times and how I could assign different COM ports with it.

Yes, this is possible. The ble-com-setup script creates always the BLE <> COM9 pair and does nothing if it already exists, but this is only for convenience.
You can create more port pairs manually, just go to your `Program Files (x86)/com0com` dir and either use command line `setupc.exe` `install` or the graphical `setupg.exe` 'Add pair' button, if you prefer that. The PortNames can be chosen arbitrarily at least for the ble-serial side, but I would recommend to use something between COM1-COM9 for the external side, because I noticed many other applications are incompatible with different naming schemes. So for example BLE2 <> COM8 should be fine.

After that run ble-serial mutiple times, I have described how this works for Linux in #52. Default cmd has no real background job feature though, instead multiple cmd windows or Windows Terminal with tabs should do the trick. Also make sure to specify the right port for every additional instance, so in this example `--port BLE2`.

Jakeler, 2022-02-24:
Closing cause it's solved.

closed 2022-02-24 by Jakeler