Jakeler/ble-serial #53: macOS Monterey breaks the package entirely

manielli, 2021-12-14:
**Describe the bug**
A clear and concise description of what the bug is.
Installation using pip or pip3 happens successfully. ble-scan doesn't return anything. As soon as downgraded to macOS Big Sur, bee-serial works again. Tried on multiple systems and laptops.

```
ble-scan doesn't return anything. Only says Started scan and finished scan.
````

**Setup (please complete the following information):**
- OS: [Linux Distribution (e.g. Ubuntu, Arch, etc), Windows Build] macOS Monterey 12.0.1 as well as 12.1
- Bluetooth Hardware: [e.g. HM-10, Intel AX200 Bluetooth Adapter] tried 2019 MacBook Pro 16in Intel as well as 2021 MacBook Pro M1 Max 16in
- BlueZ Version: [Output of `bluetoothctl -v`, Linux only]
- Python Version: [from `python3 -V`] ```python 3.8.9```
- ble-serial and dependency versions: [from `pip3 list`] 

```
pip3 list
Package                        Version
------------------------------ -------
ble-serial                     2.5.0
bleak                          0.13.0
coloredlogs                    15.0.1
humanfriendly                  10.0
pip                            20.2.3
pyobjc-core                    8.1
pyobjc-framework-Cocoa         8.1
pyobjc-framework-CoreBluetooth 8.1
pyobjc-framework-libdispatch   8.1
setuptools                     49.2.1
six                            1.15.0
wheel                          0.33.1
```

brunorodd, 2021-12-14:
Me too help! Thanks in advance!

mani-jafari, 2021-12-15:
We have tried using the ```ble-serial --dev <service_uuid> --write-uuid <write-uuid> --read-uuid <read-uuid> --mtu 187 --port /tmp/ttyBLE``` as well and it doesn't connect either. We suspect the encrypted uuids we discovered using ble-scan with macOS Big Sur may have changed from Big Sur to Monterey and that's why the above mentioned command doesn't work maybe? regardless thanks for your help, we'll keep an eye on those github issues referenced

Jakeler, 2021-12-15:
@mani-jafari BLE UUIDs are not encrypted and are the same, no matter the OS. Actually someone suggested to use the very powerful nRF Connect app on Android to find the IDs out.
What you are trying to do does not work, because scanning based on service discovery is not implemented currently. You need to manually install the bleak develop branch and also change connection code in ble-serial, to make explicit use of `BleakScanner`, add an aditional service parameter and pass it through. At runtime you have to specify both the service UUIDs and device address then, because the service is not unambiguous, there could be many devices with same services.

mani-jafari, 2021-12-15:
What is presented as BLE UUIDs as a result of scans on iOS and macOS are not the same for a specific device. The same device scanned with ```ble-serial```'s ```ble-scan``` command gives a different BLE UUID than what is scanned on iOS with libraries like ```react-native-ble-plx```. Android for instance presents true MAC Addresses when a scan is performed with ```react-native-ble-plx``` library, but macOS/iOS does not show the true MAC Address (public one) and only show the (random) MAC addresses. ```ble-serial```'s ```ble-serial``` command on Linux can use Mac Addresses to connect directly to the device but macOS requires the BLE UUID that is return as part of the ```ble-scan``` command. I wrongly used the word encrypt but I meant the random resolvable addresses:

https://www.novelbits.io/bluetooth-address-privacy-ble/

Thanks for the instructions, will give it a try.

dejanovicnenad, 2022-01-24:
@Jakeler Is there any fix coming for this? I'm on macOS Monterey and I'm unable to scan or connect to the BLE device.

[image]

[image]

Jakeler, 2022-01-28:
Yes, you can expect at least some solution, since bleak added the service uuid workaround with [0.14](https://github.com/hbldh/bleak/releases/tag/v0.14.0). 
I just did not have the time to implement it so far, might be able to do it next week, will give you guys an update here then.

SpudGunMan, 2022-01-31:
looks like osx12.3 has fixes reference https://github.com/hbldh/bleak/issues/720#issuecomment-1025017418

Jakeler, 2022-02-12:
I have implemented the service based scanning now, you can install the branch with:
```
pip install https://github.com/Jakeler/ble-serial/archive/refs/heads/mac-scan-workaround.zip
```

It includes a new parameter:
```
  -s SERVICE_UUID, --service-uuid SERVICE_UUID
                        The service used for scanning of potential devices (default: None)
```
So if you specify a existing service like this:
```
ble-scan -s 0000ffe0-0000-1000-8000-00805f9b34fbe
```
It should find devices. Same for the main:
```
ble-serial -s 0000ffe0-0000-1000-8000-00805f9b34fb -v -t 30 ...
```
Note: It is still recommended to add the `-d`/`--dev` ID, because there could be multiple devices providing the same services.

Let me know if it works for you.
Also thanks @SpudGunMan for the hint. Would be nice if someone could already try the macOS 12.3 (beta) and verify that it indeed works without service UUID again and solves the issue.

PR #54 (2022-02-20): Service based scanning (macOS 12 workaround)
closes #53

Allows to use the service UUID for scanning and connection, so the device address can be optional, also on Linux.
closed 2022-02-20 by Jakeler