Jakeler/ble-serial #55: Port is open / Device or ressource is busy

Kiritsu, 2022-02-21:
Hello,

Sorry to post this here but I'm new with all of this.

I've been trying to connect to an ELM327 device through bluetooth to serial on a Raspberry Pi 4. Everything is working fine, it propertly create a slave on `/tmp/ttyBLE` but I can't seem to do anything with it since it yells that the port is already open or that the device is busy by ble-serial (check with fuser).

Is there anything I have missed out or not understood?

Thanks 🙏

Jakeler, 2022-02-22:
Hi @Kiritsu,
not sure which ELM327 (as there are many clones) you are using, but if I remember correctly these only support Bluetooth 2.0 (classic). If that is the case ble-serial won't work, because it is specifically made for Bluetooth 4 and 5 (low energy), `rfcomm` is the right tool instead then. 
It would be helpful if you fill out the bug report template, especially post the full verbose log.

If it indeed supports 4+ and the bluetooth side works then your application might have problem with the emulated serial port. What software did you try to open the port with? For a simple read test you could try just `$ cat /tmp/ttyBLE`.

Also `/tmp/ttyBLE` is only a symlink, the real device file is in `/dev/pts/n` (the number at the end is dynamically assigned, you see it in the log), you might have more luck if you use that path directly.

Kiritsu, 2022-02-22:
Hello, 

Thank you for your answer! Since the ELM327 appeared named `OBDBLE`. I first tried to use `rfcomm` but after a few hours of trying multiple things to connect/bind everything properly, it lead to the conclusion that it is incompatible because it is bluetooth low energy.

I tried to open the port with cat and screen commands, which both yelled the issue mentioned before. However, it seemed to be a kind of permission issue since putting `666` permissions to the device file "fixed" the issue.

I think the friend I am working on this with managed to pass some data through it successfully... which would mean that my issue is fixed. 

However, and I don't know if it's related, but I have tried to open the port from a .NET 5 app using Microsoft's Serial implementation, which resulted in an IO exception `Inappropriate ioctl for device`. I'm not sure whether it's just a problem with the .NET 5 framework or anything else.

Stacktrace: 
```
System.IO.IOException: Inappropriate ioctl for device
   at System.IO.Ports.SerialStream.set_DtrEnable(Boolean value)
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
```

However, I'm supposed to upgrade and try to use the .NET 6 framework instead, they probably have fixed a few things... hopefully.

Since it also could be the Raspberry Bluetooth's hardware fault (read that in some of the articles I went in), I'll buy and try to use another Bluetooth device on the Raspberry.

I'll get back to you tomorrow or in two days with the full bug report template (since I can't try anything right now).

Thank you for your help tho!

Kiritsu, 2022-02-23:
Hello,

After upgrading the app I've built to use `System.IO.Ports` version 6, and making sure we had the right permissions with the serial port opened by `ble-serial` has fixed everything. We managed to make it work properly.

Thanks a lot for your help, and thanks a lot for your work, because it is amazing.