Jakeler/ble-serial #61: Serial error when connecting to Maxim Integrated MAXREFDES103

earubin, 2022-06-26:
Hi. I am trying to subscribe to the GATT data of the MAXREFDES103 eval product from Maxim/ADI. I am able to connect through the native GUI but I receive an error when attempting to connect to the device via ble-serial v. 2.6. I am on Windows 10 and have onboard bluetooth.

`Started deep scan of 80:6F:B0:DA:24:C5

Found device 80:6F:B0:DA:24:C5: HSP2SPO2_3_4.6 (out of 1)
SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2):  ['read']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4):  ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 6):  ['read']
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 8): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 9):  ['indicate']
SERVICE 00001523-1212-efde-1523-785feabcd123 (Handle: 11): Nordic LED Button Service
     CHARACTERISTIC 00001027-1212-efde-1523-785feabcd123 (Handle: 12):  ['read', 'write']
     CHARACTERISTIC 00001011-1212-efde-1523-785feabcd123 (Handle: 14):  ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 16): Client Characteristic Configuration
SERVICE 0000180d-0000-1000-8000-00805f9b34fb (Handle: 17): Heart Rate
     CHARACTERISTIC 00002a37-0000-1000-8000-00805f9b34fb (Handle: 18):  ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 20): Client Characteristic Configuration

Completed deep scan of 80:6F:B0:DA:24:C5

Finished BLE scan

C:\Users\...>ble-serial -d 80:6F:B0:DA:24:C5
23:11:29.175 | INFO | ble_interface.py: Receiver set up
23:11:29.176 | ERROR | main.py: Unexpected Error: SerialException("could not open port '\\\\\\\\.\\\\BLE': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)")
23:11:29.177 | WARNING | main.py: Shutdown initiated
23:11:29.178 | INFO | main.py: Shutdown complete.`

I am not sure what causes this error in the third to last line. Can someone help me out?

Jakeler, 2022-06-26:
The error means the com0com setup is not correct. 
If you run `ble-com-setup` the new window should stay open in any case, so something is not working right there. Do you get the popup (UAC) asking for admin privileges? Otherwise you could open the cmd as administrator and run `ble-com-setup` there.

If all does not work you can do the commands from the script manually, again cmd as administrator, one line after another:
```
cd "C:/Program Files (x86)/com0com/"
setupc.exe list
setupc.exe install PortName=COM9 PortName=BLE
setupc.exe list
```
Let me know what you get.

earubin, 2022-06-28:
Even when I run as administrator, the window only pops up for less than a second. And I ran those commands and got the correct output, but I am still having the issue where I get `ERROR | main.py: Unexpected Error: SerialException("could not open port '\\\\\\\\.\\\\BLE': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)")`

Jakeler, 2022-06-30:
So what exact output did you get? Also what do you get when you try it again?
Did you reboot in between? This is sometimes necessary.
You could also try to open the port with other programs, for example [ScriptCommunicator](https://sourceforge.net/projects/scriptcommunicator) works: in Settings double click in the port field and put in `\\.\BLE` - close and connect. Just to verify if the problem is the com0com installation or ble-serial.

Jakeler, 2022-07-02:
Also #29 might be related, did you already disable secure boot?

earubin, 2022-07-02:
No I did not, That got rid of the weird path issue with `\\.\BLE`. Unfortunately, I now have an issue with the device not having a write characteristic, which I can probably solve on my own. Thanks for your help.

Jakeler, 2022-07-03:
See the README, especially the section about finding an connecting a device, there is also a list. 
Only option for writing on this device is the `LED Button Service`. I am not sure if that is what you want to do with it. 
For just reading the heart rate or something you might also disable writing with `--permit ro`, note that you anyway have to specify the read characteristic, as this is non standard profile.

I am closing this again, since it is getting off topic and the original issue is solved. Feel free to open a new discussion thread if you still have questions about characteristics etc.

closed 2022-07-03 by Jakeler