Jakeler/ble-serial #65: BLE serial link for multiple devices

KamiK2K, 2022-08-02:
Hi, 

I need to use this script and tools to provide multiple COM ports for multiple BLE devices. I tried modifying the ble-serial packages to make this possible but I had no luck. Is there a way to use modify this tool to provide multiple OCM ports for multiple BLE devices (one for each device obviously)?

Thank you in advance,
Kami

Jakeler, 2022-08-02:
Hi Kami,
this is already possible, without modifications. Just run ble-serial mutiple times, I have described how this works for Linux in  https://github.com/Jakeler/ble-serial/discussions/52#discussioncomment-1782799. On Windows it is only slightly different.

- Default cmd has no real background job feature, instead multiple cmd windows or Windows Terminal with tabs should do the trick.
- You can create more com0com port pairs manually, just go to your `Program Files (x86)/com0com` and either use command line `setupc.exe` `install` or the graphical `setupg.exe` 'Add pair' button, if you prefer that. The PortNames can be chosen arbitrarily at least for the ble-serial side, but I would recommend to use something between COM1-COM9 for the external side, because I noticed many other applications are incompatible with different naming schemes. So for example BLE2 <> COM8 should be fine.

Also make sure to specify the right port for every additional instance, so in this example `--port BLE2`.

I should probably add a section to the readme, cause this question comes up quite regularly.

KamiK2K, 2022-08-03:
I appreciate your help and support Jake, I worked and it is perfect. This is a very useful tool to use.

Just a quick question, I am assuming that it is possible to establish more than two BLE<->COM links right?

I am using two BLE devices right now and I was wondering if it is possible to establish up to five BLE<->COM port links.

Sincerely,
Kami

Jakeler, 2022-08-03:
Of course you can have >2, the other discussion was also about several devices, I was just using 2 as example.

Only limiting factor is the bluetooth baseband layer, which uses a Active Member Address (AMA, 3 bit), so max. 8 values. Since address zero is always occupied by the host, you should be able to connect up to 7 devices.

commit e9f82213 (2022-09-11): Add multi device connection instructions to readme

explains and closes #65
closed 2022-09-11 by Jakeler