Jakeler/ble-serial #73: Bluetooth Connection Timeout.

chrisjaxy, 2022-11-25:
- OS: Ubuntu 20.04
- Bluetooth Hardware: MTK-7921
- BlueZ Version: 5.53
- Python Version: 3.8.10
- ble-serial and dependency versions: ble-serial：2.6.1，bleak：0.19.5

**Additional Context**
I refer to the instructions to connect to BLE Bluetooth, such as ESP32, WLT8016, but I find that the Bluetooth connection is always disconnected. I grab the logs with hcidump and btmon, and it shows Connection Timeout. This rarely happens when I use bluetoothctl to connect to a Bluetooth device. I wonder if there is any difference between these two connections?

Jakeler, 2022-11-26:
It really should not make a difference, both bluetoothctl and ble-serial/bleak utilize the bluez dbus API. They should send practically the same commands.
Is there is specific time when the connection breaks? did you also try enabling notifications/indications via bluetoothctl, maybe that causes issues?
Otherwise you could check with the super verbose mode `-vv` or what happens on the bus `dbus-monitor --system "destination='org.bluez'"`.

chrisjaxy, 2022-11-28:
According to your method, I found these in the ble-serial log

> DEBUG | manager.py: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_01_00_0A_DA_67_25): ['org.bluez.Device1', {'ServicesResolved': <dbus_fast.signature .Variant ('b', False)>, 'Connected': <dbus_fast.signature.Variant ('b', False)>}, []]’

In dbus-monitor, there is no special log for disconnection.

> method return time=1669600087.902222 sender=org.freedesktop.DBus -> destination=:1.526 serial=344 reply_serial=22785