Jakeler/ble-serial #85: autoconnect issue raspberry pi

AndreasMulzer, 2023-05-15:
Hi,
first of all, a big thank you. This stuff works flawless. Expect one little thing. The autoconnect sript returns a failure, when I start it.
The terminal says the following:

python ble-serial/helper/ble-autoconnect.py -h
  File "ble-serial/helper/ble-autoconnect.py", line 16
    async def run_tool(conf_section: dict):
            ^
SyntaxError: invalid syntax

I really have no idea on how to fix this. Can someone please asist? Thanks

Jakeler, 2023-05-16:
Hi @AndreasMulzer , 
what distro/OS are you running on the rpi? 
The error looks like an unsupported python version (as it has problems with the async function definition).
You can check with `python -V`, let me know if it says 2.7... then you have to run it with `python3 ble-serial/helper/ble-autoconnect.py` and it should work.
I could probably improve the documentation on that part a bit.

AndreasMulzer, 2023-08-13:
Yes, this was the key. Thanks a lot!

closed 2023-08-13 by AndreasMulzer