Jakeler/ble-serial #87: Script fails completely

byteptr, 2023-07-07:
I have Raspberry pi with Raspbian:
Linux rpi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

I installed ble-serial trying both options: 
Option 1: 
 pip3 install ble-serial
 Option 2: 
 git clone https://github.com/Jakeler/ble-serial.git
 cd ble-serial/
pip install -r requirements.txt

On both cases If I try to launch python -m ble_serial -d XX:XX:XX:XX:XX:XX I have the followong error 

Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/repos/ble-serial/ble_serial/__main__.py", line 1, in <module>
    from ble_serial.main import launch
  File "/home/pi/repos/ble-serial/ble_serial/main.py", line 4, in <module>
    from ble_serial.ports.tcp_socket import TCP_Socket
  File "<fstring>", line 1
    (host=)
         ^
SyntaxError: invalid syntax

I can't understand what is going wrong here, but it seems to fails because it doesn't parse correctly the command line parameters

byteptr, 2023-07-07:
I even tried directly and got same error 

ble-serial -d 6C:79:B8:1E:33:14
Traceback (most recent call last):
  File "/usr/local/bin/ble-serial", line 5, in <module>
    from ble_serial.main import launch
  File "/usr/local/lib/python3.7/dist-packages/ble_serial/main.py", line 4, in <module>
    from ble_serial.ports.tcp_socket import TCP_Socket
  File "<fstring>", line 1
    (host=)
         ^
SyntaxError: invalid syntax

byteptr, 2023-07-07:
I even tried directly and got same error 

ble-serial -d XX:XX:XX:XX:XX:XX
Traceback (most recent call last):
  File "/usr/local/bin/ble-serial", line 5, in <module>
    from ble_serial.main import launch
  File "/usr/local/lib/python3.7/dist-packages/ble_serial/main.py", line 4, in <module>
    from ble_serial.ports.tcp_socket import TCP_Socket
  File "<fstring>", line 1
    (host=)
         ^
SyntaxError: invalid syntax

byteptr, 2023-07-07:
Hi guys, 
I take a look to your code, then I commented the 5th line on your file tcp_socket.py  and it seems this  was the culprit of crash, I'm on python 3.7

`class TCP_Socket(ISerial):
    def __init__(self, host: str, port: int, mtu: int):
        logging.debug(f'{host=}:{port=}, {mtu=}') <---  Line 5  ?????  
        self.host = host
        self.port = port
        self.mtu = mtu
        self.connected = False
`

Now it seems to work, but I have another error

`20:48:09.060 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/pts/2
20:48:09.062 | INFO | ble_interface.py: Receiver set up
20:48:09.636 | INFO | ble_interface.py: Trying to connect with 6C:79:B8:1E:33:14: PEAKTECH3442 MM
20:48:10.419 | INFO | ble_interface.py: Device XX:XX:XX:XX:XX:XX connected
20:48:10.423 | ERROR | main.py: Unexpected Error: AssertionError("No characteristic in builtin write list ['0000ff01-0000-1000-8000-00805f9b34fb', '0000ff02-0000-1000-8000-00805f9b34fb', '0000ffe1-0000-1000-8000-00805f9b34fb', '6e400002-b5a3-f393-e0a9-e50e24dcca9e', '6e400003-b5a3-f393-e0a9-e50e24dcca9e'] found!\n                    Please specify one with -w/--write-uuid, see also --help")
20:48:10.425 | WARNING | main.py: Shutdown initiated
20:48:10.426 | INFO | linux_pty.py: Serial reader and symlink removed
20:48:12.568 | WARNING | ble_interface.py: Device 6C:79:B8:1E:33:14 disconnected
20:48:12.569 | INFO | ble_interface.py: Stopping Bluetooth event loop
20:48:12.571 | INFO | ble_interface.py: Bluetooth disconnected
20:48:12.572 | INFO | main.py: Shutdown complete.
`

byteptr, 2023-07-07:
Hi guys,
I take a look to your code, then I commented the 5th line on your file tcp_socket.py and it seems this was the culprit of crash, I'm on python 3.7
`class TCP_Socket(ISerial):
    def __init__(self, host: str, port: int, mtu: int):
        logging.debug(f'{host=}:{port=}, {mtu=}') <--- Line 5
        self.host = host
        self.port = port
        self.mtu = mtu
        self.connected = False
`
Now it seems to work,

Jakeler, 2023-08-24:
Hi @byteptr, thanks for reporting, this is actually a valid bug.
ble-serial version 2.7 broke python 3.7 compatibility with the TCP modules and I forget to specify it correctly in the metadata, will raise the min. requirements in the next release. Self-documenting expressions are a python 3.8 feature. Shouldn't be a problem with lastest Raspberry Pi OS.

commit a429060f (2023-08-27): Drop python 3.7 compatibility, add 3.11

fixes #87
closed 2023-08-27 by Jakeler