Jakeler/ble-serial #92: BLE virtual serial ports with characteristics implementing flow control

Yoda-6, 2023-10-13:
**Is your feature request related to a problem? Please describe.**
I've been using this excellent package for a few months to help test a BLE virtual serial link between a vendors BLE SIP module and a PC. The intention is to replace a wired serial connection. The module vendor has recently updated their SIP module firmware to require the use of the MODEM_IN and MODEM_OUT characteristics which mimic the serial hardware CTS/RTS handshaking signals (not sure if MODEM_IN/OUT is their nomenclature of generic). This change to the module FW breaks the compatibility with this package, data is sent from the PC side to the module ok, but traffic from the module to the PC is not received.

**Describe the solution you'd like**
Would it be possible to modify ble-serial to support these additional flow-control characteristics? Thanks in advance.

**Additional context**
Deep scan of the SIP BLE module running the new (not working) FW
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed ['indicate']
         DESCRIPTOR 00000000-0000-1000-8000-00805f9b34fb (Handle: 4): None
SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 5): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 6): Device Name ['read', 'write']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 8): Appearance ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 10): Peripheral Preferred Connection Parameters ['read']
     CHARACTERISTIC 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 12): Central Address Resolution ['read']
SERVICE 1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0 (Handle: 14): Unknown
     CHARACTERISTIC f7bf3564-fb6d-4e53-88a4-5e37e0326063 (Handle: 15): Unknown ['write']
SERVICE 569a1101-b87f-490c-92cb-11ba5ea5167c (Handle: 17): Unknown
     CHARACTERISTIC 569a2001-b87f-490c-92cb-11ba5ea5167c (Handle: 18): Unknown ['write-without-response', 'write']
     CHARACTERISTIC 569a2000-b87f-490c-92cb-11ba5ea5167c (Handle: 20): Unknown ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 22): Client Characteristic Configuration
     CHARACTERISTIC 569a2003-b87f-490c-92cb-11ba5ea5167c (Handle: 23): Unknown ['write-without-response', 'write']
     CHARACTERISTIC 569a2002-b87f-490c-92cb-11ba5ea5167c (Handle: 25): Unknown ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 27): Client Characteristic Configuration

And a different module (same type) running the older (working) FW
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed ['indicate']
         DESCRIPTOR 00000000-0000-1000-8000-00805f9b34fb (Handle: 4): None
SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 5): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 6): Device Name ['read', 'write']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 8): Appearance ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 10): Peripheral Preferred Connection Parameters ['read']
     CHARACTERISTIC 00002aa6-0000-1000-8000-00805f9b34fb (Handle: 12): Central Address Resolution ['read']
SERVICE 1d14d6ee-fd63-4fa1-bfa4-8f47b42119f0 (Handle: 14): Unknown
     CHARACTERISTIC f7bf3564-fb6d-4e53-88a4-5e37e0326063 (Handle: 15): Unknown ['write']
SERVICE 569a1101-b87f-490c-92cb-11ba5ea5167c (Handle: 50): Unknown
     CHARACTERISTIC 569a2001-b87f-490c-92cb-11ba5ea5167c (Handle: 51): Unknown ['write-without-response', 'write']
     CHARACTERISTIC 569a2000-b87f-490c-92cb-11ba5ea5167c (Handle: 53): Unknown ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 55): Client Characteristic Configuration
     CHARACTERISTIC 569a2003-b87f-490c-92cb-11ba5ea5167c (Handle: 56): Unknown ['write-without-response', 'write']
     CHARACTERISTIC 569a2002-b87f-490c-92cb-11ba5ea5167c (Handle: 58): Unknown ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 60): Client Characteristic Configuration

Jakeler, 2023-10-14:
Hey @Yoda-6, we already had this in discussions/ideas a while ago: https://github.com/Jakeler/ble-serial/discussions/43
Flow control support would be certainly nice to have, but right now I neither have hardware that supports it nor the time required for a deep dive in the topic.
The `CHARACTERISTIC 569a2003-b87f-490c-92cb-11ba5ea5167c` seems to be from [Laird Tech](http://cdn.lairdtech.com/home/brandworld/files/Application%20Note%20-%20Using%20VSP%20with%20smartBASIC.pdf). Also the BL600 is 10 years old and with BLE 4.0 very outdated now. Not sure what exactly would make sense now, probably something from their Lyra 24 series.
That being written, if you don't care about actual flow control it's probably very simple to hack. Sending already works, ModemOut is just ignored by ble-serial. Nothing is received because the module waits for a 1 in ModemIn, so if you just set that byte with `bluetoothctl` etc. after the initial connection it should happily transfer data, no need to change it afterwards.

Yoda-6, 2023-10-16:
Thanks for the rapid and comprehensive response. Apologies for not finding the pre-existing discussion on this before posting. I'll have a look into using the hack you suggested.