Jakeler/ble-serial #94: ble-autoconnect problem on Windows 7

snhirsch, 2024-01-06:
**Describe the bug**
A clear and concise description of what the bug is.

When I start ble-serial at the command line my application is able to open the COM7 port and communicate properly with the remote device.  However, if I let autoconnect handle it the application claims the port is in use and cannot open it.

**Log messages**

When started manually at command prompt:
```
C:\Users\hirsch>ble-serial -d 38:AB:41:F8:D7:C7
16:09:44.480 | INFO | ble_interface.py: Receiver set up
16:09:44.907 | INFO | ble_interface.py: Trying to connect with 38:AB:41:F8:D7:C7: Extron 2
16:09:46.479 | INFO | ble_interface.py: Device 38:AB:41:F8:D7:C7 connected
16:09:46.479 | INFO | ble_interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
16:09:46.479 | INFO | ble_interface.py: Found notify characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
16:09:46.557 | INFO | main.py: Running main loop!
16:10:32.096 | WARNING | main.py: Shutdown initiated
16:10:32.096 | INFO | windows_com0com.py: Stopping RX+TX loop
16:10:35.406 | WARNING | ble_interface.py: Device 38:AB:41:F8:D7:C7 disconnected
16:10:35.406 | INFO | ble_interface.py: Stopping Bluetooth event loop
16:10:35.422 | INFO | ble_interface.py: Bluetooth disconnected
16:10:35.422 | INFO | main.py: Shutdown complete.
```
When started from autoconnect:
```
C:\Users\hirsch\ble-autoconnect.py:51: FutureWarning: This method will be removed in a future version of Bleak. Use the detection_callback of the BleakScanner constru
  scanner.register_detection_callback(detection_callback)
C:\Users\hirsch\ble-autoconnect.py:36: FutureWarning: BLEDevice.rssi is deprecated and will be removed in a future version of Bleak, use AdvertisementData.rssi instea
  logging.info(f'{device.address} = {device.name} (RSSI: {device.rssi})')
[INFO] 38:AB:41:F8:D7:C7 = None (RSSI: -80)
[INFO] Found 38:AB:41:F8:D7:C7 in config!
[INFO] 38:AB:41:F8:D7:C7 = Extron 2 (RSSI: -80)
[INFO] Found 38:AB:41:F8:D7:C7 in config!
[INFO] ['ble-serial', '--dev', '38:AB:41:F8:D7:C7', '--port', 'COM7']
16:14:01.243 | INFO | ble_interface.py: Receiver set up
16:14:01.493 | INFO | ble_interface.py: Trying to connect with 38:AB:41:F8:D7:C7: Extron 2
16:14:03.057 | INFO | ble_interface.py: Device 38:AB:41:F8:D7:C7 connected
16:14:03.057 | INFO | ble_interface.py: Found write characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
16:14:03.072 | INFO | ble_interface.py: Found notify characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (H. 17)
16:14:03.135 | INFO | main.py: Running main loop!
```

**Setup (please complete the following information):**
- OS: [Linux Distribution (e.g. Ubuntu, Arch, etc), Windows Build]  Windows 7
- Bluetooth Hardware: [e.g. HM-10, Intel AX200 Bluetooth Adapter] HM-10
- Python Version: [from `python3 -V`]  3.8.6 (newest that will run on Win 7)
- ble-serial and dependency versions: [from `pip3 list`]  

```
C:\Users\hirsch>pip list
Package           Version
----------------- -------
async-timeout     4.0.3
ble-serial        2.7.1
bleak             0.21.1
bleak-winrt       1.2.0
coloredlogs       15.0.1
humanfriendly     10.0
pip               20.2.1
pyreadline3       3.4.1
pyserial          3.5
setuptools        49.2.1
typing-extensions 4.9.0
```

**Additional Context**
Tricks for reproducing the issue? Results from other things you have already tried? Ideas what might be the cause, etc.

Jakeler, 2024-01-06:
Hey @snhirsch, I see the run from cmd was without `--port` argument, that means it uses the default `BLE` internal port. To make it work simply change your autoconnect.ini to `port = BLE` or completely remove the port lines. 
It needs to be always the other side of a port pair, so if your app connects to COM7 then specifying the same is wrong. You can see this in com0com configuration.
It's different on Linux/macOS, where any valid path works and ports get dynamically created by ble-serial the need of additional tools.

snhirsch, 2024-01-06:
Oh, ouch...  You are of course correct .  I must have looked at that ini file a dozen times and never put it together.

Working fine now.   Thanks so much for your efforts on this!  Your code solved a major problem for me.  Keep up the great work.

closed 2024-01-06 by snhirsch