Jakeler/ble-serial #95: MAC address is case sensitive

snhirsch, 2024-01-06:
This one is simple, but caused some issues until I figured it out.  The ble-serial application does not work correctly when the alpha characters in the MAC address are passed in lower case.

Jakeler, 2024-01-06:
Interesting, I just tested it on Linux and could not reproduce it: upper, lower and even mixed case all worked.
Assume it is WinRT specific, will test it on Windows 10/11 later. You use Windows 7 and the same software versions you mentioned in #94, right?

snhirsch, 2024-01-06:
Yes, same environment as the other issue.

Jakeler, 2024-01-06:
Still unconfirmed on W10, with Python 3.8, 3.10 and 3.12 it all worked. Same package versions.

Actually W7 is unsupported by the [bleak](https://github.com/hbldh/bleak) dependency:
> Supports Windows 10, version 16299 (Fall Creators Update) or greater

Idk how this can function [at all.](https://github.com/hbldh/bleak/issues/364) Though I don't have a W7 test system at hand.

snhirsch, 2024-01-06:
Well, it is just a minor annoyance.  With full logging on I see a bunch of warnings about deprecated API usage that I believe comes from 'bleak'.  I can only assume they haven't yet pulled it completely.  Now I know never to update it!  

I'll close the issue.  Thanks again!

closed 2024-01-06 by snhirsch