Jakeler/ble-serial #96: Autoconnect w/ multiple devices?

snhirsch, 2024-01-08:
I'm not sure if this is a bug or the intended behavior.  I have two HM-10 devices active and have created a section in the autoconnect.ini file for each.  One gets associated with the BLE virtual port and the other with BLE2.  I have created both port pairs:  BLE<-->COM7 and BL2<-->COM8.  Both work individually if I connect manually, but with the autoconnect.py script only one of them (seemingly at random) is ever connected.  After the connect it announces that it's entering the main loop and that's it!  Shouldn't it continue the scan and also pickup the other device?  Or, am I misunderstanding the purpose of the program?

Jakeler, 2024-01-09:
It's intended behavior for the basic implementation, the [scanner stops](https://github.com/Jakeler/ble-serial/blob/main/helper/ble-autoconnect.py#L17) while a device is connected and only resumes afterwards. There was never much interest in the autoconnect tool, so a multi device feature was never requested there and I didn't touch it for almost 3 years.
That being said, the architecture with asyncio tasks should work well with multiple subprocesses, it's probably enough to remove the scanner stop line. Some extensions are needed just to make this more practical. 
1. connected devices can show up again as scan result, to avoid multiple connections with the same device it should track connected devices and filter the scans against it.
2. currently logs from ble-serial are passed into the same terminal, this would get messy and confusing with multiple runs in parallel. An idea is to prefix each line with a process/device id. Or always capture it and write into files. Or use some external terminal multiplexer like tmux, multiple terminal tabs/windows etc. this would be nice, but very OS specific, also makes (1) more difficult with the wrapper not immediately returning or not passing the exit code.

Jakeler, 2024-12-09:
I went with the prefixing line approach. There is now a `name` config option so you can set it to whatever you like.

It required a few more changes because connections fail when the scanner runs continuously, the scanner has to stop for some time (`-t` parameter) to let ble-serial startup. Then it listens again for other devices. Successfully tested with two at the same time.

You can try it with the new 3.0.0 release.

commit 0d1dea5f (2024-12-09): Implement autoconnect with multiple devices
using new --name arg for clearer combined logs
closes #96

also update deprecated bleak and asyncio api
closed 2024-12-09 by Jakeler