Jakeler/ble-serial #97: Add hint in documentation

atxbyea, 2024-02-12:
Note: for general support questions, please use the discussions tab. This is the template for bugs in ble-serial.
Make sure you have completely read the [README](https://github.com/Jakeler/ble-serial/blob/master/README.md), searched through the [issues](https://github.com/Jakeler/ble-serial/issues?q=+) and [discussions](https://github.com/Jakeler/ble-serial/discussions).

**Describe the bug**
A clear and concise description of what the bug is.

**Log messages**
The full output of `ble-serial -v` (with your usual parameters like `-d`).
If it is related to BLE chracteristics please also run `ble-scan -d` with you device address.

Put the result into code blocks (wrap with triple backticks) to get proper formatting:
```
your log could be here
```

**Setup (please complete the following information):**
- OS: [Linux Distribution (e.g. Ubuntu, Arch, etc), Windows Build]
- Bluetooth Hardware: [e.g. HM-10, Intel AX200 Bluetooth Adapter]
- BlueZ Version: [Output of `bluetoothctl -v`, Linux only]
- Python Version: [from `python3 -V`]
- ble-serial and dependency versions: [from `pip3 list`]

**Additional Context**
Tricks for reproducing the issue? Results from other things you have already tried? Ideas what might be the cause, etc.

Jakeler, 2024-02-12:
Hey, thanks for sharing the infos for this device! 
So you got it working with the instructions using deep scan and found the correct uuid. Idk is there anything specific that could be improved in the existing docu/readme? 

We could maybe add it to the autodetect list, but I like to have a specification from the manufacturer, employee or at least multiple users confirming the same uuid.

Otherwise this issue can be closed, others can still find it by searching for Airconsole BLE etc.

atxbyea, 2024-02-13:
Ah, sorry, would be my assumption that since the Airconsole BLE device is the only one with serial port I've seen it would be a common device in the userbase of this repository.

Just keeping it as a closed issue would also make it searchable, which is great, if any other users can confirm their UUIDs that would make it easy to implement in the code I guess.

closed 2024-02-13 by Jakeler