Jakeler/ble-serial #98: TIOCMGET ioctl failed

chris4git, 2024-03-04:
Hi,
I have an application who expect a serial device and do several ioctl calls on it. Unfortunately it fails when using ble-serial  : 
```
openat(AT_FDCWD, "/tmp/ttyBLE", O_RDWR|O_NOCTTY|O_NONBLOCK) = 41
ioctl(41, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(41, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(41, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(41, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(41, TIOCMGET, 0x7ffddbaa7314)     = -1 ENOTTY (Ioctl() inapproprié pour un périphérique)
```
TIOCMGET is used to get the status of modem bits.

Is it possible to implement this ioctl ?

Jakeler, 2024-03-07:
Hey @chris4git, this is not possible with the pty that ble-serial automatically creates, it's just not supported in linux for this interface.
If you are ok with compiling and loading a small kernel module though take a look at tty0tty - linux null modem emulator. It implements these commands and you get port pairs in `/dev/tnt*`, so your app should be able to connect with one side. Use `socat` to connect the other side with the port from ble-serial. 
Note that while the calls are implemented there, the returned values do not represent the actual hardware. So it's a problem if modem bits are actually used, also writing with TIOCMSET has not effect on the ble module. It can work if this is only some startup check.

chris4git, 2024-03-07:
thanks you for your suggestions. I'll look at tty0tty. 
The returned values do not matter because only Rx and Tx are used. I don't understand why there is a TIOCMSET call...

closed 2024-05-14 by Jakeler