Jakeler/ble-serial #99: Connection failed: Device not found

pH-Valiu, 2024-03-11:
I get the error: "Bluetooth connection failed. Device not found", although the device clearly exists as I managed to do ble-scan where it popped up and even managed to do a deep scan on it. 

Can anyone help me?
[image]

Jakeler, 2024-03-11:
Try with `--address-type random`

Jakeler, 2024-03-12:
The process to establish a connection can be different, depending on the type it first has to cryptographically derive the actual address. See: https://novelbits.io/bluetooth-address-privacy-ble/
In your case it's the random static type (0xFF is ..11). 

I could probably make this parameter optional though, because the ble stack already knows which one it is when it detects the device/advertisement.

pH-Valiu, 2024-03-13:
Thank you for the answer

closed 2024-03-13 by pH-Valiu