Jakeler/ble-serial #100: Windows 10/11 needing Secure Boot off

pH-Valiu, 2024-03-16:
Hi, I am currently using your script and I really like it. 
Unfortunately I am stuck to using Windows and I really do not favor to use this script in production with the PC in Secure Boot off.

Is there any way to circumvent this? Or rather what is the reason for the need to be disabled in the first place?

Thank you in advance :)

Jakeler, 2024-03-17:
Hey, the only reason to have secure boot off is the serial port emulation with com0com, but ble-serial itself has no hard dependency on it.
Microsoft did enforce the requirements for kernel mode drivers more. Following their certification process and regularly renewing the code signing certs is expensive - just not feasible for the com0com open source project.

But there are other solutions, I briefly mention this in the readme:
> Otherwise there exist multiple proprietary serial port emulators, these should work too. Just manually create a pair that includes a port named BLE.

For example here are a few listed: https://www.virtual-serial-port.org/articles/top-6-virtual-com-port-apps/
Note: this site/article is from eltima and they sell one of the options. They also [explicitly mention](https://wiki.eltima.com/user-guides/vspd/vspd-9-versions.html) that they got it signed to work with secure boot since 2019.
When you search for "virtual serial port emulator" there is also some freeware (closed source). I didn't test any of them and don't know how trustworthy it is, as usual do you own research before loading or buying.

Also please share you experience if you tried one, would be interesting for other windows users.

pH-Valiu, 2024-03-17:
Thank you very much for the insight. :)

So I tried it by utilizing "Virtual Serial Port Driver" from [https://www.virtual-serial-port.org/](url)
They have a 14 day trial (or ca. 140$ paid version).

Just creating a Pair with one named "BLE" made your scripts work even in Secure Boot mode ON
Here it also didn't matter whether the Secure Boot was on: "Microsoft only" or "Microsoft and 3rd party CA"

So that is really great news 👍 even if the pricing is really hefty and it only being an application you need to launch once yourself, instead of a SDK, which you could also get, to integrate into your project. That option is available aswell (they didn't disclose the price though 😂 )

Currently my project (Java) works by running an internal command line and calling your python scripts. Therefore it requires the user to have already installed python AND the scripts, which is kinda unhandy. 

Any idea where I would start when wanting to transform your python script into Java?

pH-Valiu, 2024-03-18:
There are even more options out there. 
Like you said, the only requirement is being able to create a virtual serial port pair.

Apart from com0com, I only found other really free option create the pair (without the need to have the application always up and running).

That option is VSP Manager fom K5FR [https://k5fr.com/DDUtilV3wiki/index.php/VSP_Manager](url). 
The downside of it being him only sending the software to you if you are a HAM radio amateur. (Luckily for me, I am)

closed 2024-05-13 by Jakeler