Jakeler/ble-serial #101: macos-Raspberry Serial comms possible?

seamusdemora, 2024-04-06:
Sorry to drop in like this, but your site showed up in a search, and since it looks promising, I wanted to ask a question:

Is it possible to set up serial comms (as in tty) between a macos & a raspberry pi via Bluetooth?  I have read that something called "Serial Port Profile" is disabled in macos, and this makes BT serial comms impossible. However, just fiddling around earlier I was able to make a BT connection between my macos (Macbook) and a Raspberry Pi. After making the connection, I checked `/dev` on my Mac and found some interesting items:  `cu.Bluetooth-Incoming-Port`, `cu.rpi3a`, `tty.Bluetooth-Incoming-Port` & `tty.rpi3a`. 

I tried `screen /dev/tty.rpi3a 19200`, but got nothing.

Jakeler, 2024-04-06:
Hey, you have to look at Bluetooth 2 (classic / legacy) and version 4+ separately. Serial Port Profile (SPP) is part of the legacy stack. Small embedded devices usually support only a single protocol, but afaik both the raspberry and macbooks are all dual mode.

These device files on the system are related to SPP and should theoretically still work, if the other side speaks the same protocol. On the rpi/linux you would need some sort of "SPP server" to get data through. For example: https://superuser.com/questions/142288/how-to-configure-linux-to-act-as-a-bluetooth-rfcomm-spp-server but idk if this still works, since it's 13 years old.

If you want to utilize more modern BLE 4+ then ble-serial server mode could be interesting for you too. I didn't have time to solve all quirks to consider it release ready, but according to the feedback it already works well, so if you want to try: https://github.com/Jakeler/ble-serial/issues/47#issuecomment-986692540 Install this on one side and connect with normal ble-serial client on the other.

seamusdemora, 2024-04-06:
@Jakeler 
Thanks for all of that!  I'll try this just as soon as I work through this other BT issue I've got - I've committed to get an RPi working to play "rain sounds" through a BT speaker for my wife... helps her sleep :)

closed 2024-05-13 by Jakeler