Jakeler/ble-serial #102: Windows 10 Unable to connect to the peripheral after discovery

syettapu123, 2024-05-17:
Note: for general support questions, please use the discussions tab. This is the template for bugs in ble-serial.
Make sure you have completely read the [README](https://github.com/Jakeler/ble-serial/blob/master/README.md), searched through the [issues](https://github.com/Jakeler/ble-serial/issues?q=+) and [discussions](https://github.com/Jakeler/ble-serial/discussions).

**Describe the bug**
On windows 10   BLE serial command is able to identify the device and unable to make the connection completion. 
Whereas same device works on ubuntu 20.0, and serial console is working. 

**Log messages**
The full output of `ble-serial -v` (with your usual parameters like `-d`).
C:\Users\syettapu>ble-serial -vv -d 56:34:12:EF:CD:AB
09:05:37.631 | DEBUG | main.py: Running: Namespace(verbose=2, timeout=5.0, adapter='hci0', mtu=20, device='56:34:12:EF:CD:AB', addr_type='public', service_uuid=None, write_uuid=None, read_uuid=None, mode='rw', filename=None, binlog=False, port='BLE', tcp_host='127.0.0.1', tcp_port=None)
09:05:37.631 | DEBUG | proactor_events.py: Using proactor: IocpProactor
09:05:37.631 | INFO | ble_interface.py: Receiver set up
09:05:38.033 | DEBUG | scanner.py: Received E0:60:89:00:1A:06: .
09:05:38.048 | DEBUG | scanner.py: Received E0:60:89:00:1A:06:
09:05:38.048 | DEBUG | scanner.py: Received AC:23:3F:AC:99:4B: .
09:05:38.048 | DEBUG | scanner.py: Received AC:23:3F:AC:99:4B: .
09:05:38.048 | DEBUG | scanner.py: Received AC:23:3F:AD:66:50: .
09:05:38.164 | DEBUG | scanner.py: Received E0:60:89:00:50:CA: .
09:05:38.164 | DEBUG | scanner.py: Received E0:60:89:00:50:CA:
09:05:38.171 | DEBUG | scanner.py: Received 68:D8:8C:D8:79:F2: .
09:05:38.171 | DEBUG | scanner.py: Received 68:D8:8C:D8:79:F2: .
09:05:38.171 | DEBUG | scanner.py: Received AC:23:3F:AC:99:52: .
09:05:38.171 | DEBUG | scanner.py: Received 05:3A:89:99:67:14: .
09:05:38.171 | DEBUG | scanner.py: Received AC:23:3F:77:41:04: .
09:05:38.271 | DEBUG | scanner.py: Received 50:32:37:CC:3E:32: .
09:05:38.287 | DEBUG | scanner.py: Received 30:F4:44:9D:3E:B7: .
09:05:38.387 | DEBUG | scanner.py: Received 30:F4:44:9D:3E:B7: .
09:05:38.465 | DEBUG | scanner.py: Received 50:32:37:CC:3E:32: .
09:05:38.465 | DEBUG | scanner.py: Received 50:32:37:CC:3E:32: .
09:05:38.471 | DEBUG | scanner.py: Received AC:23:3F:69:2B:EC: .
09:05:38.471 | DEBUG | scanner.py: Received AC:23:3F:69:2B:EC: .
09:05:38.471 | DEBUG | scanner.py: Received 36:29:0C:81:F8:10: .
09:05:38.471 | DEBUG | scanner.py: Received AC:23:3F:AD:66:50: .
09:05:38.503 | DEBUG | scanner.py: Received 6B:2B:1B:03:E2:99: .
09:05:38.519 | DEBUG | scanner.py: Received 38:03:B2:8B:88:C8: .
09:05:38.519 | DEBUG | scanner.py: Received AC:23:3F:AD:66:4E: .
09:05:38.519 | DEBUG | scanner.py: Received AC:23:3F:AD:66:4E: .
09:05:38.634 | DEBUG | scanner.py: Received AC:23:3F:AC:99:58: .
09:05:38.650 | DEBUG | scanner.py: Received AC:23:3F:AD:66:61: .
09:05:38.665 | DEBUG | scanner.py: Received 25:0B:4B:2F:7D:03: .
09:05:38.672 | DEBUG | scanner.py: Received 56:34:12:EF:CD:AB: EVT ECHO.
09:05:38.672 | DEBUG | scanner.py: 19 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
09:05:38.688 | INFO | ble_interface.py: Trying to connect with 56:34:12:EF:CD:AB: EVT ECHO
09:05:38.688 | DEBUG | client.py: Connecting to BLE device @ 56:34:12:EF:CD:AB
09:05:38.719 | ERROR | main.py: Bluetooth connection failed: Device with address 56:34:12:EF:CD:AB was not found.
09:05:38.719 | WARNING | main.py: Shutdown initiated
09:05:38.719 | INFO | windows_com0com.py: Stopping RX+TX loop
09:05:38.719 | INFO | main.py: Shutdown complete.

If it is related to BLE chracteristics please also run `ble-scan -d` with you device address.
Started deep scan of 56:34:12:EF:CD:AB

Found device 56:34:12:EF:CD:AB: None (out of 1)
SERVICE 00001801-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Attribute Profile
     CHARACTERISTIC 00002a05-0000-1000-8000-00805f9b34fb (Handle: 2): Service Changed ['indicate']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 4): Client Characteristic Configuration
SERVICE 00001800-0000-1000-8000-00805f9b34fb (Handle: 5): Generic Access Profile
     CHARACTERISTIC 00002a00-0000-1000-8000-00805f9b34fb (Handle: 6): Device Name ['read', 'write']
     CHARACTERISTIC 00002a01-0000-1000-8000-00805f9b34fb (Handle: 8): Appearance ['read']
     CHARACTERISTIC 00002a04-0000-1000-8000-00805f9b34fb (Handle: 10): Peripheral Preferred Connection Parameters ['read']
SERVICE 6e400001-b5a3-f393-e0a9-e50e24dcca9e (Handle: 12): Nordic UART Service
     CHARACTERISTIC 6e400002-b5a3-f393-e0a9-e50e24dcca9e (Handle: 13): Nordic UART RX ['write-without-response', 'write']
     CHARACTERISTIC 6e400003-b5a3-f393-e0a9-e50e24dcca9e (Handle: 15): Nordic UART TX ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 17): Client Characteristic Configuration
Completed deep scan of 56:34:12:EF:CD:AB

Put the result into code blocks (wrap with triple backticks) to get proper formatting:
```
09:05:38.672 | DEBUG | scanner.py: Received 56:34:12:EF:CD:AB: EVT ECHO.
09:05:38.672 | DEBUG | scanner.py: 19 devices found. Watcher status: <BluetoothLEAdvertisementWatcherStatus.STOPPED: 3>.
09:05:38.688 | INFO | ble_interface.py: Trying to connect with 56:34:12:EF:CD:AB: EVT ECHO
09:05:38.688 | DEBUG | client.py: Connecting to BLE device @ 56:34:12:EF:CD:AB
09:05:38.719 | ERROR | main.py: Bluetooth connection failed: Device with address 56:34:12:EF:CD:AB was not found.
09:05:38.719 | WARNING | main.py: Shutdown initiated
09:05:38.719 | INFO | windows_com0com.py: Stopping RX+TX loop
09:05:38.719 | INFO | main.py: Shutdown complete
```

**Setup (please complete the following information):**
- OS: Windows Build 19045.4412
- Bluetooth Hardware: Thinkpad-p15v intel wifi6 ax201+ blue 5 vpro
- Python Version: Python 3.12.3
- ble-serial and dependency versions:
Package                                                 Version
------------------------------------------------------- -------
ble-serial                                              2.7.1
bleak                                                   0.22.1
coloredlogs                                             15.0.1
humanfriendly                                           10.0
pip                                                     24.0
pyreadline3                                             3.4.1
pyserial                                                3.5
winrt-runtime                                           2.0.1
winrt-Windows.Devices.Bluetooth                         2.0.1
winrt-Windows.Devices.Bluetooth.Advertisement           2.0.1
winrt-Windows.Devices.Bluetooth.GenericAttributeProfile 2.0.1
winrt-Windows.Devices.Enumeration                       2.0.1
winrt-Windows.Foundation                                2.0.1
winrt-Windows.Foundation.Collections                    2.0.1
winrt-Windows.Storage.Streams                           2.0.1
**Additional Context**

Jakeler, 2024-05-21:
Looks similar to: https://github.com/Jakeler/ble-serial/issues/99 - does this solve it for you too?

Jakeler, 2024-12-11:
Stale, no reaction - closing

closed 2024-12-11 by Jakeler