Jakeler/ble-serial #104: Error: No characteristic with specified write UUID found

Matejvvo, 2024-05-26:
Hi, I am sort of a beginner when it comes to bluetooth and BLE. I want to connect to serial over BLE on LEGO Education SPIKE Prime Controller. I need to connect to the serial to access repl and micro python. I am able to connect to the controller and message with it over BLE using the documentation and example code attached bellow. However I cannot make it work with this tool.

I found the UUID characteristics for the controller here: https://lego.github.io/spike-prime-docs/connect.html. 
```
Service	0000FD02-0000-1000-8000-00805F9B34FB
RX		0000FD02-0001-1000-8000-00805F9B34FB
TX		0000FD02-0002-1000-8000-00805F9B34FB
```

I tried this command and I was not successful. (I also tried switching rx and tx for w/r and I tried differnet -w/-r/-s combinations)
```bash
$ ble-serial -d 395880E9-DC16-617C-BAE9-C9FAFBD3C96B -w 0000FD02-0001-1000-8000-00805F9B34FB -r 
```

This is the log of the error.
```bash
0000FD02-0002-1000-8000-00805F9B34FB -s 0000FD02-0000-1000-8000-00805F9B34FB
16:57:10.598 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/ttys003
16:57:10.598 | INFO | ble_interface.py: Receiver set up
16:57:10.706 | INFO | ble_interface.py: Trying to connect with 395880E9-DC16-617C-BAE9-C9FAFBD3C96B: HUB_FLL01
16:57:12.977 | INFO | ble_interface.py: Device 395880E9-DC16-617C-BAE9-C9FAFBD3C96B connected
16:57:12.977 | ERROR | main.py: Unexpected Error: AssertionError('No characteristic with specified write UUID 0000FD02-0001-1000-8000-00805F9B34FB found!')
16:57:12.978 | WARNING | main.py: Shutdown initiated
16:57:12.978 | INFO | linux_pty.py: Serial reader and symlink removed
16:57:12.979 | WARNING | ble_interface.py: Device 395880E9-DC16-617C-BAE9-C9FAFBD3C96B disconnected
16:57:12.979 | INFO | ble_interface.py: Stopping Bluetooth event loop
16:57:12.979 | INFO | ble_interface.py: Bluetooth disconnected
16:57:12.979 | INFO | main.py: Shutdown complete.
```

Can you please help me figure out how to make this work? Or did I misunderstand completely, and this tool is not meant to be used for this purpose?

Thanks for your time and your advice!

Jakeler, 2024-05-26:
I looked at the example code and the device should work for sure with ble-serial.
Please post the log with `-v` (and your usual parameters like `-d/-w/-r`).
Since it is related to BLE chracteristics also run `ble-scan -d` with your device address.

Matejvvo, 2024-05-26:
Thanks for the quick response.

Here is the output of deep scan:
```
Started deep scan of 395880E9-DC16-617C-BAE9-C9FAFBD3C96B

Found device 395880E9-DC16-617C-BAE9-C9FAFBD3C96B: HUB_FLL01 (out of 1)
SERVICE 0000fd02-0000-1000-8000-00805f9b34fb (Handle: 7): Vendor specific
     CHARACTERISTIC 0000fd02-0001-1000-8000-00805f9b34fb (Handle: 8): Unknown ['write-without-response']
     CHARACTERISTIC 0000fd02-0002-1000-8000-00805f9b34fb (Handle: 10): Unknown ['notify']
         DESCRIPTOR 00002902-0000-1000-8000-00805f9b34fb (Handle: 12): Client Characteristic Configuration

Completed deep scan of 395880E9-DC16-617C-BAE9-C9FAFBD3C96B
```

And here is log with `-v`:
```
17:42:25.179 | DEBUG | main.py: Running: Namespace(verbose=1, timeout=5.0, adapter='hci0', mtu=20, device='395880E9-DC16-617C-BAE9-C9FAFBD3C96B', addr_type='public', service_uuid='0000FD02-0000-1000-8000-00805F9B34FB', write_uuid='0000FD02-0001-1000-8000-00805F9B34FB', read_uuid='0000FD02-0002-1000-8000-00805F9B34FB', mode='rw', filename=None, binlog=False, port='/tmp/ttyBLE', tcp_host='127.0.0.1', tcp_port=None)
17:42:25.179 | DEBUG | selector_events.py: Using selector: KqueueSelector
17:42:25.181 | INFO | linux_pty.py: Port endpoint created on /tmp/ttyBLE -> /dev/ttys003
17:42:25.181 | INFO | ble_interface.py: Receiver set up
17:42:25.291 | INFO | ble_interface.py: Trying to connect with 395880E9-DC16-617C-BAE9-C9FAFBD3C96B: HUB_FLL01
17:42:25.839 | INFO | ble_interface.py: Device 395880E9-DC16-617C-BAE9-C9FAFBD3C96B connected
17:42:25.839 | ERROR | main.py: Unexpected Error: AssertionError('No characteristic with specified write UUID 0000FD02-0001-1000-8000-00805F9B34FB found!')
17:42:25.839 | WARNING | main.py: Shutdown initiated
17:42:25.839 | INFO | linux_pty.py: Serial reader and symlink removed
17:42:25.840 | WARNING | ble_interface.py: Device 395880E9-DC16-617C-BAE9-C9FAFBD3C96B disconnected
17:42:25.841 | INFO | ble_interface.py: Stopping Bluetooth event loop
17:42:25.841 | INFO | ble_interface.py: Bluetooth disconnected
17:42:25.841 | INFO | main.py: Shutdown complete.

```

Jakeler, 2024-05-26:
Ok, I think the issue is simply the uppercase id, try it with `-w 0000fd02-0001-1000-8000-00805f9b34fb -r 0000fd02-0002-1000-8000-00805f9b34fb`

Matejvvo, 2024-05-26:
Yes lowercase works, thanks very much for your help.

closed 2024-05-26 by Matejvvo