Jakeler/ble-serial #106: Unexpected Error: FileExistsError

noduritoto, 2024-07-30:
Note: for general support questions, please use the discussions tab. This is the template for bugs in ble-serial.
Make sure you have completely read the [README](https://github.com/Jakeler/ble-serial/blob/master/README.md), searched through the [issues](https://github.com/Jakeler/ble-serial/issues?q=+) and [discussions](https://github.com/Jakeler/ble-serial/discussions).

**Describe the bug**
I met a error about 'Unexpected Error: FileExistsError'
_How can I fix it?_

**Log messages**
The full output of `ble-serial -v` (with your usual parameters like `-d`).
If it is related to BLE chracteristics please also run `ble-scan -d` with you device address.

Put the result into code blocks (wrap with triple backticks) to get proper formatting:
$ble-serial -d 12345678989

your log could be here

$ble-serial -d 12345678989
14:08:06.889 | ERROR | main.py: Unexpected Error: FileExistsError(17, 'File exists')
14:08:06.889 | WARNING | main.py: Shutdown initiated
14:08:06.890 | INFO | main.py: Shutdown complete.

**Setup (please complete the following information):**
- OS: MacOS Sonoma
- Bluetooth Hardware: [e.g. HM-10, Intel AX200 Bluetooth Adapter]
- BlueZ Version: [Output of `bluetoothctl -v`, Linux only]
- Python Version: [from `python3 -V`]
- ble-serial and dependency versions: [from `pip3 list`]

**Additional Context**
Tricks for reproducing the issue? Results from other things you have already tried? Ideas what might be the cause, etc.

MrCocoDev, 2024-07-31:
I'm not the author, but if you use the `--port` parameter to change the port you can circumvent this error. I am not sure why the virtual port isn't being cleaned up automatically.

Jakeler, 2024-07-31:
Hi, you can delete the file if you are sure no ble-serial is running, in the console simply: `rm /tmp/ttyBLE`

It gets auto cleaned on a usual shutdown, but maybe the process was hard killed before, then it can't run the cleanup routine.

The tip from @MrCocoDev for using a different path should work as well.

MrCocoDev, 2024-07-31:
I think if you add:

```py
                try:
                    self.uart = UART(args.port, loop, args.mtu)
                except OSError as e:
                    if "FileExistsError(17, 'File exists')" in repr(e):
                        raise ValueError(
                            f"Port {args.port} already exists. "
                            f"Please specify a new port or clean "
                            f"up the old port, if it is safe to do so."
                        ) from e 
```

You could add a lot of clarity around this issue. I happened to figure it out from an unrelated SO and a good guess, but I imagine a lot of users will just give up in frustration.

Either way, thanks for such an awesome library!

Jakeler, 2024-11-25:
Good idea @MrCocoDev , thanks for the contribution!
It's added now in the UART class.

commit 583866bd (2024-11-25): Improve error message for already existing port symlinks

showed only a generic FileExistsError on main otherwise, which is not clear for users

fixes #106
closed 2024-11-25 by Jakeler